emuflight/EmuFlight #806: Target Request  HAkRCF722D

torju01, 2022-05-05:
Vendor/Model:  Hakrc F722 AIO 40A  (F7226 aio)
target name:    HAkRCF722D

link to the manufactur: maybe http://www.hikrc.com/ but card is not listed
link to a vendor (pics and details):  https://www.getfpv.com/hakrc-aio-board-brushless-f722-fc-40a-2-6s-blheli-s-esc-w-external-usb.html
  
Unified Target:   https://github.com/betaflight/unified-targets/blob/master/configs/default/HARC-HAKRCF722D.config

Thank you Veryy much

torju01, 2022-05-05:
Hi Beau Brewski, as it is possible than Hakrc look Matek architecture, i will test MATEKF722 target tomorow !
and I come back to you for result  :-)

BeauBrewski, 2022-05-06:
> Hi Beau Brewski, as it is possible than Hakrc look Matek architecture, i will test MATEKF722 target tomorow ! and I come back to you for result :-)

I checked the BF config files, and this is not compatible with `MATEKF722`. Will need a new target for this one.

BeauBrewski, 2022-05-06:
Needs tested!
Check gyro orientation, motor arrangement, flash, osd, vtx control, everything else if possible please update here

PROPS OFF FOR SAFETY!

[EmuFlight_0.4.0_HAKRCF722D_Build_fbd081b8a.zip](https://github.com/emuflight/EmuFlight/files/8642671/EmuFlight_0.4.0_HAKRCF722D_Build_fbd081b8a.zip)

torju01, 2022-05-07:
Hello Beau,
it flies :-)
gyro orientation set to 180 on roll, 180 on pitch, -45 on yaw
motor arrangement, osd, vtx are all good
barometer non tested
many thanks for your help

BeauBrewski, 2022-05-07:
Does this board have blackbox and does that appear to be recognized?

torju01, 2022-05-07:
yes 16mo, just check, there is some emu data in it (emuf_all.bbl)

Moksh-2000, 2022-07-07:
I have same board and i can test this?  Here some information that comes with box 

[image]

[image]

closed 2022-07-29 by nerdCopter