emuflight/EmuFlight #810: TMOTORF7

StuartD89, 2022-05-23:
T-Motor F7 HD 
Target: TMOTORF7

https://store.tmotor.com/goods.php?id=935

[Apex - RCInpower 2306.5 1880kv.txt](https://github.com/emuflight/EmuFlight/files/8758099/Apex.-.RCInpower.2306.5.1880kv.txt)

nerdCopter, 2022-05-24:
this one doesn't work? 
[image]

StuartD89, 2022-05-24:
Hello,

No it does not, I tried it hoping that it would still work but it just locked my flight controller out, I had to start it up in boot mode to flash betaflight back on it and on betaflight it uses the TMOTORF7 target.

[image]<https://user-images.githubusercontent.com/56646290/170037185-98616eb9-fb95-4f61-ae11-27d61e494d21.png>

BeauBrewski, 2022-05-24:
I have an analog build that uses this target. Flashed `TMOTORF7` (the 0.3.4 version), gyros and everything worked. Flashed `TMTR-TMOTORF7` (0.4.0 version) gyros and everything worked. Even flashed `TMOTORF7V2` and gyros and booting worked (however other resources are definitely wrong as incorrect target). 

Unable to recreate above problem. 
Being HD build should not change anything except how you set up ports and OSD.

StuartD89, 2022-05-25:
I will try to flash again later, when I flashed it the first time the computer wouldn’t even recognise the USB port. I used boot mode to get it back.

BeauBrewski, 2022-05-25:
While not always necessary depending on program, it is always good practice to unplug/replug after software changes (whether flashing ESCs or firmware). 

Let us know if any other issues appear!

Closing issue for now.

closed 2022-05-25 by BeauBrewski