emuflight/EmuFlight #812: Feature Request/Target/Bugfix: MATEKF411 ICM20689 gyro support

BeauBrewski, 2022-05-25:
Per Goldensniperss on discord, Gyro not working on Darwinfpv 15A 1-3S F411 Ultralight/Whoop AIO which uses `MATEKF411` firmware.

Per Darwinfpv website, this board can have the ICM20689 gyro.
https://darwinfpv.com/products/darwinfpv-15a-1-3s-f411-ultralight-whoop-aio

Upgrading .hex for `MATEKF411` to include ICM29689. (Previously done for `MATEKF411RX` due to upgrades to the HappyModel Mobula6 FC/`CRAZYBEEF4FR` target.)

(Note: Once BMI270 driver support is done, will need to update this target again 😞 )

BeauBrewski, 2022-05-25:
Needs tested for gyro functionality.

[EmuFlight_0.4.0_MATEKF411_Build_27fc065cc.zip](https://github.com/emuflight/EmuFlight/files/8774653/EmuFlight_0.4.0_MATEKF411_Build_27fc065cc.zip)

nerdCopter, 2022-05-27:
all the icm mpu bmi :zany_face:

nightflasher, 2022-05-27:
Downloaded the zip, flashed my DarwinFPV flight controller and it works fine!
Thank you!

nightflasher, 2022-06-06:
Okay, the gyro works for flying and that is what it should do in first place.
But: G-Force in OSD stays at 0.0, so something isn't working there.

nerdCopter, 2022-06-11:
beta4 worked good and speedy! :+1:   i had enabled logs, but no files saved upon exit.

nightflasher, 2022-07-03:
Can't be wrong, if this is related to this beta hex file

nerdCopter, 2022-07-23:
@BeauBrewski , does this need to be PR'd and merged?  
fix g-force some later date?

nerdCopter, 2022-07-23:
[image]

nightflasher, 2022-08-14:
yeah, please merge!

nightflasher, 2023-02-26:
is it possible to add the DarwinF411 target in emuFlight?

BeauBrewski, 2023-02-26:
> yeah, please merge!
> 
> updated to 0.4.1 and on the flying field: NO GYRO - hmpf

[EmuFlight_0.4.1_MATEKF411_Build_10e71c86d.zip](https://github.com/emuflight/EmuFlight/files/10834792/EmuFlight_0.4.1_MATEKF411_Build_10e71c86d.zip)

Here, give this a test and we'll work on getting it merged

nightflasher, 2023-03-01:
nope, still no gyro.

nerdCopter, 2023-03-01:
@nightflasher , do you have a Betaflight `status` output from CLI to verify the gyro?

BeauBrewski, 2023-03-01:
Oh my bad, I realize what I did wrong. Derp. 
Try it this time. Same as original hex up top, but updated to 0.4.1....for real this time.

[EmuFlight_0.4.1_MATEKF411_Build_abfbf0e9d.zip](https://github.com/emuflight/EmuFlight/files/10862283/EmuFlight_0.4.1_MATEKF411_Build_abfbf0e9d.zip)

nightflasher, 2023-03-07:
okay, now it's working.

BeauBrewski, 2023-05-21:
Working on test target for bmi270
https://github.com/emuflight/EmuFlight/pull/890