emuflight/EmuFlight #822: MATEKF722HD and MATEKF722SE not detecting Barometer

zheroz00, 2022-07-04:
**Describe the bug**
*Emu 0.4.0*  Barometer not detected on Matek F722-HD or Matek F722-SE.

**To Reproduce**
Flash to the correct target. Barometer not detected when activated.

**Expected behavior**
To see the barometer indicator in Emuflight light up when activated.

**Flight controller configuration**

```
# version
# EmuFlight / MATEKF722HD (MTKS) 0.4.0 Apr  1 2022 / 12:50:55 (40faa0ab2) MSP API: 1.51

board_name 
manufacturer_id 

# name

# resources

# mixer

# servo

# servo mix

# feature
feature -AIRMODE
feature GPS

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 0 2 115200 9600 0 115200
serial 1 0 115200 57600 0 115200
serial 3 64 115200 57600 0 115200
serial 5 1 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 900 1200 0 0
aux 1 1 5 900 1200 0 0
aux 2 13 2 1800 2100 0 0
aux 3 28 7 1800 2100 0 0
aux 4 35 6 1800 2100 0 0
aux 5 36 4 1800 2100 0 0
aux 6 40 0 1800 2100 0 0

# adjrange
adjrange 0 0 3 900 2100 12 3 0 0

# rxrange

# vtx

# rxfail

# master
set acc_calibration = 73,51,332
set mag_hardware = NONE
set mag_declination = -7570
set min_check = 1005
set serialrx_provider = CRSF
set gps_provider = UBLOX
set osd_warn_arming_disable = ON
set osd_warn_batt_not_full = ON
set osd_warn_batt_warning = ON
set osd_warn_batt_critical = ON
set osd_warn_visual_beeper = ON
set osd_warn_crash_flip = ON
set osd_warn_esc_fail = ON
set osd_warn_core_temp = ON
set osd_warn_rc_smoothing = ON
set osd_warn_dji = ON
set osd_vbat_pos = 2496
set osd_rssi_pos = 2075
set osd_flymode_pos = 2048
set osd_throttle_pos = 448
set osd_craft_name_pos = 2218
set osd_gps_sats_pos = 2080
set osd_home_dir_pos = 2095
set osd_home_dist_pos = 2061
set osd_warnings_pos = 2538
set osd_avg_cell_voltage_pos = 2464
set osd_disarmed_pos = 2251
set osd_stat_endbatt = ON

# profile
profile 0

# rateprofile
rateprofile 0

set thr_mid = 45
set thr_expo = 10
set roll_rc_rate = 125
set pitch_rc_rate = 125
set yaw_rc_rate = 90
set roll_expo = 75
set pitch_expo = 75
set yaw_expo = 75
set roll_srate = 0
set pitch_srate = 0
set yaw_srate = 0
```

**Setup / Versions**
 - Matek F722-HD and Matek F722-SE. Bought from Greathobbies in Canada.
 - DIATONE MAMBA F60_BL32 60A
 - RX = Betafpv ELRS nano 2.4.
 - VTX = DJI air unit. 
 - GPS = BN-220
 - *Wiring* Standard connections to ESC and motors. GPS on UART1. RX on UART4. DJI on UART6.

**Additional context**
I tested BF 4.3 and the barometer shows up by default.

nerdCopter, 2022-07-04:
`DPS310` not supported yet.

BeauBrewski, 2022-07-05:
See https://github.com/emuflight/EmuFlight/issues/551