emuflight/EmuFlight #826: T-MOTOR F411 LITE

liu-root, 2022-07-07:
**T-Motor F411 Lite**
* Please provide the target name. e.g. `TMOTORF411`
* Please provide vendor and model of the FC. e.g. `T-Motor Velox F411 Lite 30x30 Flight Controller`
* Please provide a link to the vendor specifications.
 e.g https://store.tmotor.com/goods.php?id=1229

**CLI dump or Unified Target**
* Please provide a CLI `dump` of **DEFAULTS** or link to unified target. : https://github.com/betaflight/unified-targets/blob/master/configs/default/TMTR-TMOTORF411.config
 
Hello team, I would like to request the support package for the above FC. Also, I can collaborate with any testing needed.

BeauBrewski, 2022-07-07:
There's going to be some delay in getting this one to you, as it has the new-ish BMI270 gyro on it and we're still working on getting that compatible with Emuflight.

BeauBrewski, 2023-05-21:
Working on test target
https://github.com/emuflight/EmuFlight/pull/890

BeauBrewski, 2023-12-07:
@nerdCopter just got another request for this.

nerdCopter, 2023-12-07:
* created by script only, did not investigate resultant files
* [EmuFlight_0.4.2_TMTR_TMOTORF411_Build_9b20fc84e_Analog.hex.zip](https://github.com/emuflight/EmuFlight/files/13600868/EmuFlight_0.4.2_TMTR_TMOTORF411_Build_9b20fc84e_Analog.hex.zip)

  - [Yes/No/Untested] hex Flash & subsequent USB connect
  - [Yes/No/Untested] Gyro 
  - [Yes/No/Untested] Accelerometer
  - [Yes/No/Untested] Is the Serial Count correct? (Ports tab UARTs)
  - [Yes/No/Untested] Motors
  - [Yes/No/Untested] Onboard ADC
  - [Yes/No/Untested] OSD (analog)
  - [Yes/No/Untested] Smart Audio controls
  - [Yes/No/Untested] Beeper/Alarm switch
  - [Yes/No/Untested] Barometer
  - [Yes/No/Untested] Blackbox  flash-chip/sdcard
  - [Yes/No/Untested] LED Strip
  - [Yes/No/Untested] GPS
  - [Yes/No/Untested] BlueTooth (if applicable)

nerdCopter, 2024-04-19:
resolved by https://github.com/emuflight/EmuFlight/pull/1036
also refer to https://github.com/emuflight/EmuFlight/issues/487

closed 2024-04-19 by nerdCopter