emuflight/EmuFlight #851: No COM port after flashing 0.4.1 TMTR_TMPACERF7_29_07_2022

B-nutze-RR, 2022-08-02:
**Describe the bug**
0.4.0 is running, but i get no COM port after flashing 0.4.1 - tested on 2 "T-Motor Pacer SINGEL Sided F7" FCs
**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

nerdCopter, 2022-08-02:
After flashing, many FC's require USB unplug & replug. this was the case with  0.4.0, 0.3.X as well.
Is this what you're describing?

B-nutze-RR, 2022-08-03:
Absolutely, i guess it's because i never experienced it with any of my Helio/Strix  setups and i now slowly add more F7/H7 FCs 😆👌

closed 2022-08-03 by B-nutze-RR