emuflight/EmuFlight #853: Target: SPEEDYBEEF745AIO

BeauBrewski, 2022-08-06:
**Vendor/Model**
`SPEEDYBEEF745AIO`

SpeedyBee F745 AIO
https://www.speedybee.com/speedybee-f745-35a-bls-25-5x25-5-aio-flight-controller/

**CLI dump or Unified Target**
https://github.com/betaflight/unified-targets/blob/master/configs/default/SPBE-SPEEDYBEE_F745_AIO.config

*Note: So far uses MPU6000 gyro, but may need to add BMI270 functionality in future.*

BeauBrewski, 2022-08-06:
Needs tested!
Check gyro orientation, motor arrangement, flash, osd, vtx control, battery readings, Bluetooth, everything else if possible please update here.

PROPS OFF FOR SAFETY!
[EmuFlight_0.4.1_SPEEDYBEEF745AIO_Build_2fda86d52.zip](https://github.com/emuflight/EmuFlight/files/9274901/EmuFlight_0.4.1_SPEEDYBEEF745AIO_Build_2fda86d52.zip)

BeauBrewski, 2022-08-09:
Per MattM on discord:
"Installed and flew perfect, was able to connect to it via my phone with speedybee and configure. The speedybee app kept crashing when I would go into the motors tab, but everything else works."

nerdCopter, 2023-09-07:
guess we duplicated https://github.com/emuflight/EmuFlight/issues/921  🙃
921 uses my script.

closed 2023-09-07 by nerdCopter