emuflight/EmuFlight #858: Target Fix: MATEKF722-PX (or maybe MATEKF722HD)

BeauBrewski, 2022-08-25:
**Vendor/Model**
MATEKSYS Flight Controller F722-PX
https://www.mateksys.com/?portfolio=f722-px#tab-id-10

**CLI dump or Unified Target**
https://github.com/betaflight/unified-targets/blob/master/configs/default/MTKS-MATEKF722HD.config

Issue:
Per Bjoern R on fb page, PX OSD is not working, as well as flash. Looking into settings.

Note: settings at bottom of page https://www.mateksys.com/?p=5159#tab-id-10

Will double check Emu target `MATEKF722PX` against BF config and INAV target.

BeauBrewski, 2022-08-25:
See also https://github.com/emuflight/EmuFlight/issues/698

BeauBrewski, 2022-08-25:
Look into flash memory support 25Q256JVEQ for Emuflight
https://github.com/iNavFlight/inav/issues/8258

BeauBrewski, 2022-08-25:
[EmuFlight_0.4.1_MATEKF722PX_Build_4695c2b18.zip](https://github.com/emuflight/EmuFlight/files/9428404/EmuFlight_0.4.1_MATEKF722PX_Build_4695c2b18.zip)

Needs tested for blackbox. 
OSD cannot be fixed at the moment.

JulioCesarMatias, 2022-08-25:
using `#define USE_FLASH_W25M` will solve the problem here on Emu, and the memory will be detected.

Now in iNav, the story is different... The drivers will have to be added, a lot will have to be modified, that is, it will be a bit of a big job

BeauBrewski, 2022-08-26:
> using `#define USE_FLASH_W25M` will solve the problem here on Emu, and the memory will be detected.

It worked 🎆 

Will update both targets and do pull requests soon to fix black box.

BeauBrewski, 2022-08-27:
[image]

BeauBrewski, 2022-08-27:
https://github.com/emuflight/EmuFlight/pull/859
https://github.com/emuflight/EmuFlight/pull/860

Close when merged. FrskyOSD pr when/if can be connected to #698

PR #859 (2022-08-27): Target Updates - MATEKF722PX - Fixed Black Box
Tested fix for Black Box on MATEKF722PX target
[image]

As well as some minor upkeep to fix a definition for SPI2 pin and remove MAX7456 osd definitions as this FC uses FrskyOSD, which is not currently supported by Emuflight. 

Merge if approved. 
Closes #858
closed 2022-08-27 by nerdCopter