emuflight/EmuFlight #861: BMI270

nerdCopter, 2022-08-31:
* No support for BMI270
* Been trying to get BMI270 code compiling for multiple weeks unsuccessfully :(

nerdCopter, 2022-08-31:
i wonder if @yangyeah is interested in this task? 
i plan to keep trying, but seems more complex than expected and doesn't look good for me.

BeauBrewski, 2022-09-01:
Ditto

DusKing1, 2022-09-02:
- If you are suffering memory issue, please use the `maximum fifo` driver
- You only need to include the `bmi270.c` or `bmi270_maximum_fifo.c` in `target.mk`

nerdCopter, 2022-09-02:
> * If you are suffering memory issue, please use the `maximum fifo` driver
> * You only need to include the `bmi270.c` or `bmi270_maximum_fifo.c` in `target.mk`

thank you kind sir, i have tried a multitude of ports, but i will "reset" and try again.

DusKing1, 2022-09-02:
> > * If you are suffering memory issue, please use the `maximum fifo` driver
> > * You only need to include the `bmi270.c` or `bmi270_maximum_fifo.c` in `target.mk`
> 
> thank you kind sir, i have tried a multitude of ports, but i will "reset" and try again.

feel free to dm me if you need more help hehe

JulioCesarMatias, 2022-09-02:
I'd like to help with that... But unfortunately I don't have any FC with the BMI270

DutchFPV, 2022-09-04:
> I'd like to help with that... But unfortunately I don't have any FC with the BMI270

I have 2 Foxeer F7 V3s that are useless to me until this gyro is supported.  Are you in the US?  I can send you one. 

 I'm two weeks away from launching a new racing frame and build spec intended for use with an Emuflight preset that's already in place.  Problem is about half of all racers are using the Foxeer F7.  So I'm a bit invested in this.  Unfortunately my coding skills are nil.  But it would be my pleasure to send you a board if you think it will help move this along.  DM me (Dutch) in the Discord group if interested.

JulioCesarMatias, 2022-09-04:
> I have 2 Foxeer F7 V3s that are useless to me until this gyro is supported. Are you in the US? I can send you one.
>
> I'm two weeks away from launching a new racing frame and build spec intended for [...]

I'm in Brazil...

DutchFPV, 2022-09-05:
Bummer.  Would probably take 2 weeks just to get to you. Was hoping you were stateside...

nerdCopter, 2022-09-19:
pinging another person (@Guidus93) (iFlight) that may be able to help or forward to potential help...

chucky-whiskers, 2022-12-26:
Has anyone made any progress on this yet?

nerdCopter, 2022-12-27:
> Has anyone made any progress on this yet?

nope.  only failed attempts.

Peck07, 2023-05-14:
Is there any update on this?
I've managed to compile the IFLIGHT_F745_AIO_V2 target based on the branch 20220726_BMI270_targets (of course with modifications)
I don't have this board, but have another type with BMI270, so that will be my next step, unless somebody already solved this :)
Sorry, pretty new to compiling Emuflight

nerdCopter, 2023-05-15:
> Is there any update on this? I've managed to compile the IFLIGHT_F745_AIO_V2 target based on the branch 20220726_BMI270_targets (of course with modifications) I don't have this board, but have another type with BMI270, so that will be my next step, unless somebody already solved this Sorry, pretty new to compiling Emuflight.
> 
> Edit: Did the changes on target BETAFPVF411 (the board I have), and gyro is working on the bench, so something is definitely not right smile Also, I must admit I've used an older version of accgyro_spi_bmi270.

you "do" or "do not" have your modifications working??. 
[image]

it is my understanding that  current Emu code-base is missing too many things for the bmi270 driver to work.

Peck07, 2023-05-15:
Compiling code to targets BETAFPVF411, IFLIGHT_F745_AIO_V2:
https://github.com/Peck07/EmuFlight/tree/20220726_BMI270_targets

WARNING: still needs to be tested in flight!

BeauBrewski, 2023-05-15:
@Peck07 
I can bench test the IFLIGHT_F745_AIO_V2.

Also have a Skystars with BMI270 I can test on, I believe its SKYSTARSF7HDPRO

closed 2023-11-01 by nerdCopter