emuflight/EmuFlight #867: Target : IFLIGHT_BLITZ_F411RX

BeauBrewski, 2022-09-19:
**Vendor/Model**
IFLIGHT_BLITZ_F411RX
BLITZ F411 1S 5A Whoop AIO Board with Built-in Receiver (BMI270)

**CLI dump or Unified Target**
https://github.com/betaflight/unified-targets/blob/master/configs/default/IFRC-IFLIGHT_BLITZ_F411RX.config

BeauBrewski, 2023-05-21:
Working on test target
https://github.com/emuflight/EmuFlight/pull/890

nerdCopter, 2024-04-19:
No plans for SPI based ELRS. :(
willing to make the target for external RX upon request.
closing this ticket.

nerdCopter, 2024-04-19:
actually, i'm unclear on this one, may support other protocols. please let me know.
`Supported Protocols: CC2500, SPI, D8, S-FHSS`
may need manual modifications of any automated target-conversion.

nerdCopter, 2024-05-29:
- product page: https://shop.iflight.com/BLITZ-F411-1S-5A-Whoop-AIO-Board-with-Built-in-Receiver-Pro1726
- unsure if ELRS is SPI based, but seems so.
- PR'd CC2500 version, but needs testing: https://github.com/emuflight/EmuFlight/pull/1068

BeauBrewski, 2024-05-29:
> * unsure if ELRS is SPI based, but seems so.
Appears target name for this target has changed since original post. Originally the IFLIGHT_BLITZ_F411RX was non-SPI ELRS and was just waiting on BMI270?

Now it appears that this board has changed to be SPI ELRS (IFLIGHT_F4SX1280), and that target also covers the CC2500 version. 

Should probably just make original target name IFLIGHT_BLITZ_F411RX as according to the BF config meets CC2500 compatibility.

nerdCopter, 2024-05-30:
> Should probably just make original target name IFLIGHT_BLITZ_F411RX as according to the BF config meets CC2500 compatibility.

cool, such is done. :pray: 
just prefer testing that one since it's a bit different than the other targets i've converted. hex is in the PR description.

PR #1068 (2024-07-12): [target] add IFLIGHT_BLITZ_F411RX - CC2500 version
- CC2500 version
- needs testing: [EmuFlight_0.4.3_IFLIGHT_BLITZ_F411RX_Build_2db3cdfc1.hex.zip](https://github.com/emuflight/EmuFlight/files/15486192/EmuFlight_0.4.3_IFLIGHT_BLITZ_F411RX_Build_2db3cdfc1.hex.zip)
- requires newer Configurator: https://github.com/emuflight/dev-master/releases/tag/20240501.162228-cfg

closes #867
closed 2024-07-12 by nerdCopter