emuflight/EmuFlight #872: X12 AIO FrSky looses signal every second

charluscha, 2022-10-17:
**Describe the bug**
X12 AIO FrSky looses signal every second in V 0.4.1
Same issue in Betaflight 4.3
I downgraded Betaflight to 4.2.2 and the issue is not present.
Is there an Emu Version to downgrade? I want to fly with Emu.

**To Reproduce**
Steps to reproduce the behavior:
bind and watch the throttle channel going down to 885 every time

**Expected behavior**
Please fix it :) 

**Flight controller configuration**
FrSky D8 SPI

**Setup / Versions**
X12 AIO 5in1 1-2s Flight Controller build-in 12A ESC + RX + Open VTX
Problem in Betaflight 4.3 and emuflight 0.4.1
Do not know, in which version there is the bug too.
In Beta 4.2.2 it works fine on two drones.

**Additional context**
ADC is off on the transmitter
The Transmitter Endpoints setting is right (1000 1500 2000)

nerdCopter, 2022-10-17:
maybe the flight-controller resources are saturated; try 4k loops (or even less)

charluscha, 2022-10-17:
Hello.
Not really. The same issue is in betaflight. In older version it works.

charluscha, 2022-10-18:
Seems to be the newest version. Happymodel knows the same issue
https://www.happymodel.cn/index.php/2022/10/18/aio-flight-controller-with-icm42688p-gyro-accelerometer-update-log/

BeauBrewski, 2022-10-19:
Emu doesn't support ICM42688p gyro. 
By lose signal, I'm assuming you mean RX signal and not Vtx signal?

charluscha, 2022-10-20:
Ah ok, thats a good point. 
Yes, it is only the rx signal. Ok, than no Emu for this board, sad.

BeauBrewski, 2022-10-21:
@charluscha Did you actually fly when you flashed CRAZYBEEF4FR and have the issue, or was it just when testing in the configurator?

charluscha, 2022-10-21:
It was at flying. In betaflight 4.2.9 it goes well. No issue.

nerdCopter, 2022-10-21:
i really don't see a reason this should happen.  there has only been 2 modifications to FrSky SPI since the birth of EmuFlight:
* https://github.com/emuflight/EmuFlight/pull/418
* https://github.com/emuflight/EmuFlight/pull/654

Both seem to be improvements, not detriments.

charluscha, 2022-10-21:
Ok, sorry, I only can tell my expierience. And happymodel writes on their site that they have rx problems so they made an own betaflight hex file for bf 4.3. 
And in betaflight github another person has same issue. But I can not say what the problem is. Maybe my FC is faulty.

BeauBrewski, 2022-10-21:
1) In betaflight CLI, type "status" and copy/paste or screen shot what it says and share here.
2) Please share your DUMP ALL from betaflight 4.2.9 and a DUMP ALL from Emuflight install.

charluscha, 2022-10-24:
Ah sorry, I did not save backups. I have so many problems with my builds, I was upset. But now I swapped my FC from my crux to my whoop and flashed it with emuflight. So I hope that will work.

closed 2024-04-18 by nerdCopter