emuflight/EmuFlight #873: Issue：The motor that pushes backwards in 3D flight mode responds incorrectly

DiehardFPV, 2022-10-18:
Recently tried the 3D flight mode, using firmware 0.4.1, there was no problem when testing at the configurator (accurate steering And the direction of thrust is correct), and the forward flight was normal, but the reverse flight would immediately roll over（it feels like there is a problem with the rotational logic of the motor when flying backwards）；But the same setup flies fine with BF firmware

**Flight controller configuration：diff data**
# version
# EmuFlight / HOWI_HOBBYWING_XROTORF7CONV (S7X2) 0.4.1 Jul 29 2022 / 13:28:00 (7bac6ddd0) MSP API: 1.51

board_name 
manufacturer_id 

# name
name CIDA_225

# resources

# mixer

# servo

# servo mix

# feature
feature -DYNAMIC_FILTER
feature 3D
feature ESC_SENSOR

# master
set gyro_sync_denom = 2
set gyro_lowpass_hz_roll = 130
set gyro_lowpass_hz_pitch = 130
set gyro_lowpass_hz_yaw = 100
set imuf_roll_q = 7000
set imuf_pitch_q = 7000
set imuf_yaw_q = 7000
set gyro_use_32khz = ON
set dynamic_gyro_notch_count = 1
set smith_predict_filt_hz = 20
set acc_calibration = -40,34,-52
set mag_hardware = NONE
set rssi_channel = 15
set rc_interp = AUTO
set rc_smoothing_input_hz = 0
set rc_smoothing_input_type = PT1
set serialrx_provider = CRSF
set blackbox_device = NONE
set dshot_idle_value = 360
set failsafe_off_delay = 20
set failsafe_throttle = 1300
set vbat_min_cell_voltage = 34
set yaw_motors_reversed = ON
set 3d_deadband_throttle = 20
set deadband = 3
set yaw_deadband = 3
set pid_process_denom = 2
set osd_rssi_pos = 2084
set osd_crsf_tx_pos = 2147
set osd_flymode_pos = 2133
set osd_vtx_channel_pos = 2100
set osd_crosshairs_pos = 2253
set osd_current_pos = 2091
set osd_craft_name_pos = 2506
set osd_altitude_pos = 2115
set osd_warnings_pos = 2441
set osd_avg_cell_voltage_pos = 2434
set osd_core_temp_pos = 2455
set osd_stat_max_spd = OFF
set osd_stat_max_alt = ON
set osd_stat_bb_no = OFF
set debug_mode = GYRO_SCALED
set cpu_overclock = 240MHZ
set vtx_channel = 5
set vtx_power = 4
set vtx_freq = 5820
set camera_control_mode = DAC

# profile
profile 0

set dterm_abg_alpha = 550
set dterm_abg_boost = 650
set dterm_abg_half_life = 250
set dterm_lowpass_hz_roll = 105
set dterm_lowpass_hz_pitch = 105
set dterm_lowpass_hz_yaw = 100
set dterm_lowpass2_hz_roll = 185
set dterm_lowpass2_hz_pitch = 185
set dterm_lowpass2_hz_yaw = 160
set iterm_relax_cutoff = 8
set iterm_relax_cutoff_yaw = 15
set pidsum_limit_yaw = 500
set throttle_boost = 6
set throttle_boost_cutoff = 25
set feathered_pids = 40
set i_decay = 7
set emu_boost = 150
set emu_boost_yaw = 125
set emu_boost_limit = 50
set emu_boost_limit_yaw = 30
set emu_gravity = 70
set axis_lock_multiplier = 7
set axis_lock_hz = 15
set p_pitch = 65
set i_pitch = 115
set d_pitch = 41
set p_roll = 61
set i_roll = 105
set d_roll = 43
set i_yaw = 105
set d_yaw = 3
set df_yaw = 50
set linear_thrust_low_output = 68
set linear_thrust_high_output = 15
set linear_throttle = ON
set mixer_laziness = ON

**hardware**
 - Flight controller ：HOBBYWING F7
 - ESC: RUSH BLADE SUPER 60A BL32.9
 - PROP: HQ J37
 - MOTOR: TM BLACK BIRD V2 1950KV
 - ESC SETUP:
[image]

 
Maybe I have some setup errors
Thanks for the answer, I really don't want to go back to BF when having 3 drones flashed EMUF

nerdCopter, 2022-10-18:
I'll let someone else answer this, but please do set `demag-compensation` = `high` for unrelated reasons.

BeauBrewski, 2022-10-19:
Have you tested it on the bench (no props please, ~~unless you ratchet strap that sucker down and test at a safe distance~~)? This will give a better understanding if its all motors, certain motors, etc.

nerdCopter, 2022-10-19:
@Quick-Flash thinks it  logic bug, but did not yet reply.  we briefly discussed in discord.

DiehardFPV, 2022-10-19:
> Have you tested it on the bench (no props please, ~unless you ratchet strap that sucker down and test at a safe distance~)? This will give a better understanding if its all motors, certain motors, etc.
> 
> Edit: Reading through old discord chat. This is a known issue and will work fine on bench then flip out in air. I believe it came about with introduction of 0.4.0. However your FC doesn't have a 0.3.x firmware version (@nerdCopter how hard would it be to get an 0.3.x version ported?). As far as I can see, there were attempts to fix, but not enough consistent feedback from test pilots to ever get it figured out.

Yes, I started testing at the configurator(props off), the motors are spinning normally and the response is correct, but it seems that when I pick up the drone, the yaw axis will twitch left and right,.  I will try to find a FC that can flash 0.3.x firmware, thanks for your reply~

Quick-Flash, 2022-10-19:
> Yes, I started testing at the configurator(props off), the motors are spinning normally and the response is correct, but it seems that when I pick up the drone, the yaw axis will twitch left and [...]

It is just the yaw direction that is wrong?

DiehardFPV, 2022-10-21:
> 

The loss of control came too quickly for me to see clearly,so i grabbed the drone with Props off to find out issues, when the drone was turned upside down, if I were to roll to the right, it should be motors 1 and 2 that spined faster, but in fact it seemed to be motors 4 and 3. My friend also had the same problem in 0.4.x, then he tested 0.3.x and flew well.

nerdCopter, 2022-10-21:
@DiehardFPV ,  @BeauBrewski suggested i port the `HOWI_HOBBYWING_XROTORF7CONV` to version 0.3.4 so i did (today).  you can now flash that version if it's the only option for now.

BeauBrewski, 2022-10-21:
@DiehardFPV its a temporary fix for now, but hopefully will get you flying. Confirmed its showing up in configurator. 👏 
[image]

DiehardFPV, 2022-10-23:
> @DiehardFPV its a temporary fix for now, but hopefully will get you flying. Confirmed its showing up in configurator. 👏 [image]

Thank you guys so much!

nerdCopter, 2022-11-10:
:thinking: , I'd be curious if 0.4.1 with `Legacy` motor-output-mixer, bypasses this bug.
`set mixer_impl =  LEGACY`, & `save`.

DiehardFPV, 2022-11-23:
> 🤔 , I'd be curious if 0.4.1 with motor-output-mixer, bypasses this bug.`Legacy`
> 
> `set mixer_impl = LEGACY`, & . or`save` [image]

I've tried it but it still doesn't work

nerdCopter, 2023-05-27:
@DiehardFPV , was this fixed with newer BlueJay firmware? or still seemingly an EmuFlight bug?

nerdCopter, 2024-03-28:
curious for any further news.  otherwise will close this ticket soon.

clsfpv, 2024-05-03:
> 对任何进一步的消息感到好奇。否则将很快关闭此票证。

Yes, my question is the same as DiehardFPV's, forgive me for not articulating it clearly. I've tried 0.4.1 and 0.4.0 and both have the same problem, but 0.4.2 and 0.3.4 still haven't tried, due to academic issues I may test it out next week when I have time, and hopefully it can be fixed, because EMU is my favorite open source firmware, and its flight feel is something BF can't bring me!

clsfpv, 2024-05-03:
> 对任何进一步的消息感到好奇。否则将很快关闭此票证。

In addition, one of my whoops had an issue where arm could not be activated when the angle mode was activated, and the OSD showed "LOAD". （LDARCf411 EMU0.4.1）

nerdCopter, 2024-05-03:
0.4.2 will not be fixed, but likely 0.3.4 works.  unfortunate that 0.3.4 doe snot have the modern filter/features.

for `LOAD`, please set pid denominator to 2 (4k loop)

clsfpv, 2024-05-03:
> 0.4.2 不会被修复，但 0.3.4 可能会工作。不幸的是，0.3.4 没有现代过滤器/功能。
> 
> 对于，请将 PID 分母设置为 2（4K 循环）`LOAD`
> 
> 如果目标LDARCF411在 0.3.4 版本中不存在，那么我可以根据要求创建它。

Yes, currently LDARC_F411 only 0.4.1, hope to use 0.3.4, thank you

clsfpv, 2024-05-03:
> [EmuFlight_0.3.4_LDARC_F411_Build_e248bee9b.hex.zip](https://github.com/emuflight/EmuFlight/files/15201834/EmuFlight_0.3.4_LDARC_F411_Build_e248bee9b.hex.zip)
> 
> 我还将添加到 0.3.4 资产中，以便 Confiurator 列出它。

Okay, thank you. I will continue to test the firmware of the EMU and provide you with some bug examples

nerdCopter, 2024-05-09:
i don't know how to configure a quad for 3D. can someone give me a diff of the 3d-specific settings, and maybe how a TX-model (edgeTX) should be setup?  then maybe i could hack around the code.  but TBH, i'm not the best for the math stuffs.
the 3D stuff seems to start here: https://github.com/emuflight/EmuFlight/blob/92c8571e716290b1c41df1881c548b5b4a2df6ee/src/main/flight/mixer.c#L399
there are other sections as well.

nerdCopter, 2024-05-21:
@ anyone
anyone willing to test a bunch of builds so that we can find which PR broke it?
(i.e i can start building a bunch of different commits until the breaking merge is found)

clsfpv, 2024-05-31:
> @clsfpv did you test 0.3.4?
I am sorry, I am studying in the high school in China so I can't take a test until next wednesday

clsfpv, 2024-05-31:
> @ anyone anyone willing to test a bunch of builds so that we can find which PR broke it? (i.e i can start building a bunch of different commits until the breaking merge is found)

I will test 0.3.4 next Wednesday🤜🏻

clsfpv, 2024-06-05:
> 你测试过 0.3.4 吗？

Hi,I've tested 0.3.4. 3D mode is work well!

tylercorleone, 2024-06-12:
Hi guys, I think that this could be the reason:
[here](https://github.com/emuflight/EmuFlight/pull/398/files#diff-c98e21d8f3cf216db2a51ba27291626147f6f48724944160963e60565ae349c3R861) the `controllerMix3DModeSign` is applied to the `yawMixVal` but then it is applied to the `rollPitchMixVal` **AND** the `yawMixVal` **again** [here](https://github.com/emuflight/EmuFlight/pull/398/files#diff-c98e21d8f3cf216db2a51ba27291626147f6f48724944160963e60565ae349c3R877), after few lines.

In the previous version the sign, that was called `motorOutputMixSign`, was applied only once to the overall `motorMix`, [here](https://github.com/emuflight/EmuFlight/pull/398/files#diff-c98e21d8f3cf216db2a51ba27291626147f6f48724944160963e60565ae349c3L690).

The fix COULD be removing the multiplication with the `controllerMix3DModeSign` from the `rollPitchMixVal`, that is the first line that I linked.

Could someone thest this hypotesis? I'm sorry but I can't.
I hope that this this will fix the problem.
Unfortunately at that PR's time we didn't test the 3D feature.

nerdCopter, 2024-06-12:
thank you so much @tylercorleone! ii will attempt.

nerdCopter, 2024-06-12:
@tylercorleone , would you mind specifying some exact lines? the diffs are "noisy" to read.

tylercorleone, 2024-06-12:
Of course, this is the line: `float yawMixVal = controllerMix3DModeSign * scaledAxisPidYaw * currentMixer[i].yaw;`

The differences in the `mixer.c` are too large and the file is collapsed, that why the links don't work properly.

We can catch up in Discord if this is not gonna work!

nerdCopter, 2024-06-12:
maybe modify `src/main/flight/mixer.c:868` to this? `        float yawMixVal = scaledAxisPidYaw * currentMixer[i].yaw;` ??

nerdCopter, 2024-06-12:
nope, it flips out when inverted.  right-side up seems to still work.

nerdCopter, 2024-06-12:
@tylercorleone , i commented in discord. i failed with multiple attempts to different lines of code. not sure what else to try.

nerdCopter, 2024-06-12:
anyone brave enough to test?
Please be extra cautious. test no props, then if brave, test prop at low altitude. stay clear of danger. i got this to briefly hover on a headless whoop, but lost control quickly.

[EmuFlight_0.4.3_HOBBYWING_XROTORF7CONV_Build_uncommitted_20240612CDT.hex.zip](https://github.com/user-attachments/files/15810636/EmuFlight_0.4.3_HOBBYWING_XROTORF7CONV_Build_uncommitted_20240612CDT.hex.zip)
[EmuFlight_0.4.3_LDARC_F411_Build_uncommitted_20240612CDT.hex.zip](https://github.com/user-attachments/files/15810638/EmuFlight_0.4.3_LDARC_F411_Build_uncommitted_20240612CDT.hex.zip)

REQUIRES Configurator: https://github.com/emuflight/dev-master/releases/tag/20240606.193851-cfg

```diff
diff --git a/src/main/flight/mixer.c b/src/main/flight/mixer.c
index 66d510837..ad4aa6c9b 100644
--- a/src/main/flight/mixer.c
+++ b/src/main/flight/mixer.c
@@ -865,13 +865,13 @@ void mixThingsUp(const float scaledAxisPidRoll, const float scaledAxisPidPitch,
     float controllerMixMin = 0, controllerMixMax = 0;
 
     for (int i = 0; i < motorCount; i++) {
-        float yawMixVal = controllerMix3DModeSign * scaledAxisPidYaw * currentMixer[i].yaw;
+        float yawMixVal = scaledAxisPidYaw * currentMixer[i].yaw;
         if (yawMixVal > yawMixMax) {
             yawMixMax = yawMixVal;
         } else if (yawMixVal < yawMixMin) {
             yawMixMin = yawMixVal;
         }
-        yawMix[i] = yawMixVal;
+        yawMix[i] = yawMixVal * controllerMix3DModeSign;
 
         float rollPitchMixVal = scaledAxisPidRoll * currentMixer[i].roll + scaledAxisPidPitch * currentMixer[i].pitch;
         if (rollPitchMixVal > rollPitchMixMax) {
@@ -879,15 +879,15 @@ void mixThingsUp(const float scaledAxisPidRoll, const float scaledAxisPidPitch,
         } else if (rollPitchMixVal < rollPitchMixMin) {
             rollPitchMixMin = rollPitchMixVal;
         }
-        rollPitchMix[i] = rollPitchMixVal;
+        rollPitchMix[i] = rollPitchMixVal * controllerMix3DModeSign;
 
-        float controllerMixVal = controllerMix3DModeSign * (rollPitchMixVal + yawMixVal);
+        float controllerMixVal = (rollPitchMixVal + yawMixVal);
         if (controllerMixVal > controllerMixMax) {
             controllerMixMax = controllerMixVal;
         } else if (controllerMixVal < controllerMixMin) {
             controllerMixMin = controllerMixVal;
         }
-        controllerMix[i] = controllerMixVal;
+        controllerMix[i] = controllerMixVal * controllerMix3DModeSign;
     }
 
     controllerMixRange = controllerMixMax - controllerMixMin; // measures how much the controller is trying to compensate
```

nerdCopter, 2024-06-18:
I think this fixes it, but it needs real-world testing: https://github.com/emuflight/dev-unstable/releases/tag/20240618.193242-hex
Can anyone test???

clsfpv, 2024-06-20:
> 我认为这可以解决它，但它需要实际测试：https://github.com/emuflight/dev-unstable/releases/tag/20240618.193242-hex 任何人都可以测试吗???

I'm still happy to try, but I don't know much about code, and I hope to provide me with a copy of the firmware directly for testing, still LDARC-F411,thanks.

clsfpv, 2024-06-20:
> 我认为这可以解决它，但它需要实际测试：https://github.com/emuflight/dev-unstable/releases/tag/20240618.193242-hex 任何人都可以测试吗???

Sorry I didn't review the firmware carefully, I've found the LDARC-F411 and I'll test it later!

clsfpv, 2024-06-21:
> 我认为这可以解决它，但它需要实际测试：https://github.com/emuflight/dev-unstable/releases/tag/20240618.193242-hex 任何人都可以测试吗???

3D mode works well!Thank you!

nerdCopter, 2024-06-21:
> 3D mode works well!Thank you!

Thank you thank you thank you, i will merge soon!  
Can you share DVR?

clsfpv, 2024-06-21:
> > 3D模式效果很好！谢谢！
> 
> 谢谢，我很快就会合并！你能分享DVR吗？

Sure,I will share it to you tomorrow!

clsfpv, 2024-06-21:
> > 3D mode works well!Thank you!
> 
> Thank you thank you thank you, i will merge soon! Can you share DVR?

Can you odd my friends to discord?I just joined discord,My discord name is Augenstern

nerdCopter, 2024-07-09:
New build: fixes both 2PASS and non-2PASS motor-output-mixers:
https://github.com/emuflight/dev-unstable/releases/tag/20240709.141832-hex (build of PR: https://github.com/emuflight/EmuFlight/pull/1069)

- Tested good without props on BLHeli32 32.9 and BlueJay 0.16.
- Tested good with props on Bluejay 0.19. I did not "fly", but it rotates each axis properly and was able to hover inverted.
- BlueJay 0.17, 0.18, 0.19 are known to have broken 3D startup power.  Seemingly fixed in BlueJay 0.21-RC1, but BlueJay developers would like BlackBox logs for further analysis.

Please test-fly and report feedback in either the [PR Link](https://github.com/emuflight/EmuFlight/pull/1069) or Discord.

closed 2024-07-12 by nerdCopter