emuflight/EmuFlight #874: [Firmware/Target Bug] BETAFPVF411RX no Gyro MPU6500 functionality

BeauBrewski, 2022-11-07:
**Describe the bug**
Target `BETAFPVF411RX` needs MPU6500 functionality added to it.

On a BetaFPV 65s that is not detecting gyro.
Per "status" on BF CLI, it is an MPU6500 gyro.

Harl-Towne, 2023-02-15:
The `BETAFPVF411` target also has the same issue for me.

nerdCopter, 2023-02-15:
please test and feedback.  if HD versions are needed, i'll have to do separately upon request. 
[EmuFlight_0.4.1_BETAFPVF411_Build_cf388beed.hex.zip](https://github.com/emuflight/EmuFlight/files/10743891/EmuFlight_0.4.1_BETAFPVF411_Build_cf388beed.hex.zip)
[EmuFlight_0.4.1_BETAFPVF411RX_Build_cf388beed.hex.zip](https://github.com/emuflight/EmuFlight/files/10743892/EmuFlight_0.4.1_BETAFPVF411RX_Build_cf388beed.hex.zip)

nerdCopter, 2023-02-19:
@Harl-Towne , any feedback on the HEX posted above?

Harl-Towne, 2023-02-21:
> @Harl-Towne , any feedback on the HEX posted above?

The [EmuFlight_0.4.1_BETAFPVF411_Build_cf388beed.hex.zip](https://github.com/emuflight/EmuFlight/files/10743891/EmuFlight_0.4.1_BETAFPVF411_Build_cf388beed.hex.zip) HEX flashes but still shows the `NO_GYRO` arming disable flag. 

The original issue mentions the MPU6500 gyro but my FC uses the BMI270. I couldn't find a BETAFPV FC with the MPU6500 so just to make sure we're on the same page I'm having issues with the [F4 1S 5A AIO Brushless Serial ELRS Flight Controller](https://betafpv.com/products/f4-1s-5a-aio-brushless-flight-controller-elrs-2-4g). Not sure if there are multiple boards with the same targets or if this board has been made with multiple gyros.

Hope the info there helps, let me know if you need anything else.

nerdCopter, 2023-02-21:
> The [EmuFlight_0.4.1_BETAFPVF411_Build_cf388beed.hex.zip](https://github.com/emuflight/EmuFlight/files/10743891/EmuFlight_0.4.1_BETAFPVF411_Build_cf388beed.hex.zip) HEX flashes but still shows the [...]

miscommunication; Emu does not support BMI270

nerdCopter, 2023-11-08:
BETAFPV411 BMI270 support resolved in master
- https://github.com/emuflight/EmuFlight/pull/930
- https://github.com/emuflight/EmuFlight/pull/937
- already tested

BETAFPVF411RX BMI270 and MPU6500 support in 
- https://github.com/emuflight/EmuFlight/pull/947
- needs testing

PR #947 (2024-03-28): [target] update BETAFPVF411RX adding BMI270 and MPU6500 and flash
* Resolves #874
* needs testing [EmuFlight_0.4.2_BETAFPVF411RX_Build_66778f03a.hex.zip](https://github.com/emuflight/EmuFlight/files/13299296/EmuFlight_0.4.2_BETAFPVF411RX_Build_66778f03a.hex.zip)
  - [Yes/No/Untested] hex Flash & subsequent USB connect
  - [Yes/No/Untested] Gyro 
  - [Yes/No/Untested] Accelerometer
  - [Yes/No/Untested] Motors
  - [Yes/No/Untested] Onboard ADC
  - [Yes/No/Untested] OSD (analog)
  - [Yes/No/Untested] Smart Audio controls
  - [Yes/No/Untested] Beeper/Alarm switch
  - [Yes/No/Untested] Barometer
  [... cut]
closed 2024-03-28 by nerdCopter