emuflight/EmuFlight #875: [TARGET REQUEST] SPEEDYBEEF7V3

C-Sharper, 2022-12-17:
**Vendor/Model**
* Target name: SPEEDYBEEF7V3
* Vendor and model: SpeedyBee F7 V3 Flight Controller
* Link to the vendor specifications:

[https://www.speedybee.com/speedybee-f7-v3-bl32-50a-30x30-stack/#Parameters](https://www.speedybee.com/speedybee-f7-v3-bl32-50a-30x30-stack/#Parameters)

**CLI dump or Unified Target**
* Link to unified targets: 
https://github.com/betaflight/unified-targets/blob/master/configs/default/SPBE-SPEEDYBEEF7V3.config

BeauBrewski, 2022-12-29:
This will need BMI270 gyro functionality. Might be a while.

nerdCopter, 2023-03-27:
> when new update ?

no one is working on BMI270 support, it was attempted but did not work.

kozirrokafaro, 2023-03-27:
> 

Ohh nooo

BeauBrewski, 2023-05-21:
@kozirrokafaro 

Update: Working on a test target; have functioning drivers finally.

C-Sharper, 2023-08-12:
Can someone cook this up for this FC so I can start testing?

nerdCopter, 2023-08-12:
[EmuFlight_0.4.2_SPBE_SPEEDYBEEF7V3_Build_de38c4938_BMI270-NORMAL-OSR4_Analog.hex.zip](https://github.com/emuflight/EmuFlight/files/12328274/EmuFlight_0.4.2_SPBE_SPEEDYBEEF7V3_Build_de38c4938_BMI270-NORMAL-OSR4_Analog.hex.zip)

no guarantees. test props-off first.
please feedback the following:
[Yes/No/Untested] hex Flash & subsequent USB connect
[Yes/No/Untested] Gyro 
[Yes/No/Untested] Accelerometer
[Yes/No/Untested] Motors
[Yes/No/Untested] Onboard ADC
[Yes/No/Untested] OSD (analog)
[Yes/No/Untested] Smart Audio controls
[Yes/No/Untested] Beeper/Alarm switch
[Yes/No/Untested] Barometer
[Yes/No/Untested] Blackbox sdcard/flash-chip
[Yes/No/Untested] LED Strip
[Yes/No/Untested] GPS

nerdCopter, 2023-08-15:
@C-Sharper , any testing done? feedback?

coffee-lord, 2023-09-03:
@nerdCopter This was my first time trying to flash EmuFlight so I'm not sure if I've done everything correctly.

[Yes] hex Flash & subsequent USB connect
[No] Gyro
[No] Accelerometer
[No] Motors (missing the last 2 beeps during initialization)
[No] Onboard ADC
[Untested] OSD (analog) (I use HDZero connected via MSP, without smart audio)
[Untested] Smart Audio controls
[Untested] Beeper/Alarm switch
[Yes] Barometer (the sensors page shows *something*, not sure if the value is actually accurate)
[No] Blackbox sdcard/flash-chip
[Untested] LED Strip
[Untested] GPS

Please let me know if you need further testing. I'm interested in checking out EmuFlight but SPEEDYBEEF7V3 is all I have at the moment.

nerdCopter, 2023-09-03:
@coffee-lord 
try this: `set gyro_to_use = FIRST`, `save`. results?
then try `set gyro_to_use = SECOND`, `save` results?

if i can get anything working, then i can get a proper HDZero build

coffee-lord, 2023-09-03:
`set gyro_to_use = FIRST` no change.

`set gyro_to_use = SECOND` the following started working:

Gyro
Accelerometer
Motors
Beeper/Alarm switch
Blackbox sdcard/flash-chip (actually it might've been working before, I forgot to switch the logging device to SD card)

Onboard ADC still doesn't seem to be working.

One more thing: There's no UART6 listed in the ports tab (Betaflight shows UART1-4 and UART6, EmuFlight only shows UART1-4). That's the UART I currently use for HDZero MSP.

nerdCopter, 2023-09-03:
thanks for report. let me investigate. i did find BF sets gyro 2 as default, which we will need to do also.

nerdCopter, 2023-09-03:
[EmuFlight_0.4.2_SPBE_SPEEDYBEEF7V3_updated_for_testing_BMI270.zip](https://github.com/emuflight/EmuFlight/files/12507288/EmuFlight_0.4.2_SPBE_SPEEDYBEEF7V3_updated_for_testing_BMI270.zip)

gyro2 should be default here. uart6 should show 🤞. unsure about ADC 🤞.

coffee-lord, 2023-09-03:
Confirmed. Gyro & stuff now work by default. ADC and HDZero OSD via UART6 also work. But now I can't access the CLI. It gets stuck on this output:

[image]

and doesn't let me type any commands. I'm using the HDZero-enabled build of the configurator, if that makes any difference.

Also EmuFlight sets the gyro update frequency (as well as PID loop frequency) to 8 kHz by default. Is that the correct frequency? Betaflight uses a fixed 3.20 kHz frequency for SPEEDYBEEF7V3. Am I supposed to change it myself? (and should I change it at all?)

coffee-lord, 2023-09-04:
CLI in the SpeedyBee app (via bluetooth) works intermittently and not for every command. `dump`, `resource` commands cause the CLI to freeze. I don't know any other way of remapping the motors.

nerdCopter, 2023-09-04:
Disable auto complete for now. 
I'll investigate.  8k is just what the configurator thinks. It's ~3.2k in hardware

coffee-lord, 2023-09-04:
I'm not sure if auto complete is the issue here. I disabled auto complete but commands such as `dump` still lock up the CLI. For example a `dump` command outputs this:

```
# dump

# version
# EmuFlight / SPBE_SPEEDYBEEF7V3 (SPBE) 0.4.2 Sep  3 2023 / 14:31:41 (136f035f2) MSP API: 1.51

board_name 
manufacturer_id 

# name
name -

# resources
resource BEEPER 1 C13
resource MOTOR 1 A15
resource MOTOR 2 B03
resource MOTOR 3 B04
resource MOTOR 4 B06
resource MOTOR 5 B07
resource MOTOR 6 B05
resource MOTOR 7 B00
resource MOTOR 8 B01
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 NONE
resource PWM 2 NONE
```

then the CLI becomes unresponsive and I can't exit the CLI mode without power cycling the FC.

I found a similar issue with Betaflight 3.5 https://github.com/betaflight/betaflight-configurator/issues/1140#issuecomment-419727985

nerdCopter, 2023-09-04:
it may be tuesday before i can dig in.
i may need `#define UART6_TX_PIN            NONE` -- i can't test/build today

BeauBrewski, 2023-09-04:
@coffee-lord 
Have you tried re-flashing? Sometimes a glitchy flash will cause issues especially in CLI.

nerdCopter, 2023-09-04:
uart4 TX needs to be `none`; and target skips uart5 in totality, maybe need TX/RX `none`; testing buiding for an FC i do have... will have a speedybee build soon for further testing

nerdCopter, 2023-09-04:
[EmuFlight_0.4.2_SPBE_SPEEDYBEEF7V3_Build_4f680cee1_BMI270_HDZero_uart4txnone_uart5none_needs_testing.hex.zip](https://github.com/emuflight/EmuFlight/files/12516219/EmuFlight_0.4.2_SPBE_SPEEDYBEEF7V3_Build_4f680cee1_BMI270_HDZero_uart4txnone_uart5none_needs_testing.hex.zip)

coffee-lord, 2023-09-04:
The CLI issue still persists.

What I've tried:

* Re-flashing the firmware a few times (@BeauBrewski)
* Use the `defaults` command (FC accepts the command and reboots but the CLI still hangs after)
* Re-flashing different versions of the firmware. I tried both the analog and HDZero versions from yesterday's builds. Only the original FW has a working CLI (this one https://github.com/emuflight/EmuFlight/issues/875#issuecomment-1676099930)
* Issuing different commands. Some work, some don't.

For example `set beeper_dshot_beacon_tone = 4` does work. But `resource`, `dump`, `diff` all make the FC unresponsive. Here's another example of  `dump` output:

```
Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# EmuFlight / SPBE_SPEEDYBEEF7V3 (SPBE) 0.4.2 Sep  4 2023 / 14:19:03 (4f680cee1) MSP API: 1.52

board_name 
manufacturer_id 

# name
name -

# resources
resource BEEPER 1 C13
resource MOTOR 1 A15
resource MOTOR 2 B03
resource MOTOR 3 B04
resource MOTOR 4 B06
resource MOTOR 5 B07
resource MOTOR 6 B05
resource MOTOR 7 B00
resource MOTOR 8 B01
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 C08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
```

nerdCopter, 2023-09-04:
@BeauBrewski , see anything wrong here? 
[SPBE_SPEEDYBEEF7V3_needsCLIHangFixed.zip](https://github.com/emuflight/EmuFlight/files/12516454/SPBE_SPEEDYBEEF7V3_needsCLIHangFixed.zip)

i'm building with this branch: `20230606_HDZero+master+ICM426+BMI270`

no need to commit target files, they will build just bu creating folder/files.

nerdCopter, 2023-09-04:
gonna rebuild with `#define USE_UART5`

nerdCopter, 2023-09-04:
[EmuFlight_0.4.2_SPBE_SPEEDYBEEF7V3_Build_4f680cee1_HDZero_BMI270_uart5NON_useUART5_for_testing.hex.zip](https://github.com/emuflight/EmuFlight/files/12516562/EmuFlight_0.4.2_SPBE_SPEEDYBEEF7V3_Build_4f680cee1_HDZero_BMI270_uart5NON_useUART5_for_testing.hex.zip)

coffee-lord, 2023-09-04:
No dice.

Just to double check that I flashed the correct version, here's the version string:

```
# EmuFlight / SPBE_SPEEDYBEEF7V3 (SPBE) 0.4.2 Sep  4 2023 / 15:27:43 (4f680cee1) MSP API: 1.52
```

I also re-checked the original "working but wrong" build and it did in fact have a working SD card. Or so it seems. Could it be that `SDCARD_DMA_*` stuff causes issues? Or `USE_FLASH`?

nerdCopter, 2023-09-04:
and the latest few do not work BBL/SDCard?

coffee-lord, 2023-09-04:
All firmware versions including the original seem to have a working SD card. And by "working SD card" I mean it shows the remaining free space on the blackbox tab and I can activate the mass storage mode and it successfully mounts the drive, etc. I haven't checked the actual blackbox logging.

nerdCopter, 2023-09-04:
[EmuFlight_0.4.2_SPBE_SPEEDYBEEF7V3_Build_4f680cee1_withoutUSEFLASH.hex.zip](https://github.com/emuflight/EmuFlight/files/12516666/EmuFlight_0.4.2_SPBE_SPEEDYBEEF7V3_Build_4f680cee1_withoutUSEFLASH.hex.zip)

[EmuFlight_0.4.2_SPBE_SPEEDYBEEF7V3_Build_4f680cee1_withoutUSEFLASH_withoutUSESDCARDSPI.hex.zip](https://github.com/emuflight/EmuFlight/files/12516667/EmuFlight_0.4.2_SPBE_SPEEDYBEEF7V3_Build_4f680cee1_withoutUSEFLASH_withoutUSESDCARDSPI.hex.zip)

coffee-lord, 2023-09-04:
Both FWs have working CLI and SDCARD :tada:

coffee-lord, 2023-09-04:
Thank you, @nerdCopter! Now I can finally remap the motors and see how this thing flies.

nerdCopter, 2023-09-05:
dropping working definitions here to avoid loss. [SPBE_SPEEDYBEEF7V3_known_working.zip](https://github.com/emuflight/EmuFlight/files/12524489/SPBE_SPEEDYBEEF7V3_known_working.zip)

cannot be merged until BMI270 support merged.

nerdCopter, 2023-09-05:
[EmuFlight_0.4.2_SPBE_SPEEDYBEEF7V3_BMI270_NormalDriver_OSR4_working.zip](https://github.com/emuflight/EmuFlight/files/12524599/EmuFlight_0.4.2_SPBE_SPEEDYBEEF7V3_BMI270_NormalDriver_OSR4_working.zip)

nerdCopter, 2023-11-08:
@coffee-lord , are you up for testing another build?  now that BMI270 is merged to master, i'm looking to PR a proper `SPEEDYBEEF7V3`.  
.hex is in PR https://github.com/emuflight/EmuFlight/pull/948

coffee-lord, 2023-11-08:
Sure!

[Yes] hex Flash & subsequent USB connect
[Yes] Gyro
[Yes] Accelerometer
[Yes] Motors
[Yes] Onboard ADC
[Untested] OSD (analog)
[Untested] Smart Audio controls
[Untested] Beeper/Alarm switch
[Yes] Barometer
[Yes] Blackbox sdcard/flash-chip
[Untested] LED Strip
[Untested] GPS
[Yes] BlueTooth (if applicable)

Everything seems to be working as expected. CLI works too.

nerdCopter, 2023-11-08:
> Sure!

thank you merged. if you need HD build, let me know.

closed 2023-11-08 by nerdCopter