emuflight/EmuFlight #878: Problems with the number of GPS satellites displayed

DiehardFPV, 2023-02-04:
I recently used the GPS with the M10 Generation chip, and I found that the 3D positioning was successful, and I got the correct altitude speed and latitude and longitude, but there was no number of satellites

[image]
-----------------------------------------------------------------------
# diff

# version
# EmuFlight / CLRACINGF7DUAL (CLR7) 0.4.1 Jul 29 2022 / 21:31:06 (3b2c3c763) MSP API: 1.52

board_name 
manufacturer_id 

# name
name Vroom

# resources
resource PINIO 1 A14

# mixer

# servo

# servo mix

# feature
feature -DYNAMIC_FILTER
feature GPS
feature TELEMETRY
feature OSD
feature ESC_SENSOR

# beeper
beeper -ON_USB

# beacon

# map

# serial
serial 1 65536 115200 115200 0 115200
serial 3 1024 115200 57600 0 115200
serial 4 2 115200 115200 0 115200
serial 5 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 2 900 1250 0 0
aux 2 2 2 1300 1700 0 0
aux 3 46 1 1700 2100 0 0
aux 4 13 4 1700 2100 0 0
aux 5 35 3 1700 2100 0 0
aux 6 36 5 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set align_gyro = CW180FLIP
set gyro_lowpass_hz_roll = 170
set gyro_lowpass_hz_pitch = 170
set gyro_lowpass_hz_yaw = 150
set imuf_roll_q = 7500
set imuf_pitch_q = 7500
set imuf_yaw_q = 7000
set imuf_w = 48
set gyro_use_32khz = ON
set dynamic_gyro_notch_count = 1
set smith_predict_filt_hz = 20
set align_acc = CW180FLIP
set acc_calibration = 18,-14,-26
set mag_hardware = NONE
set rssi_channel = 15
set rc_interp = AUTO
set rc_smoothing_input_hz = 0
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set dshot_idle_value = 350
set failsafe_off_delay = 20
set failsafe_throttle = 1300
set failsafe_procedure = GPS-RESCUE
set vbat_max_cell_voltage = 44
set current_meter = ESC
set vbat_scale = 150
set yaw_motors_reversed = ON
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_rescue_descent_dist = 50
set gps_rescue_throttle_hover = 1270
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 5
set deadband = 4
set yaw_deadband = 4
set pid_process_denom = 4
set osd_rssi_alarm = 40
set osd_alt_alarm = 120
set osd_rssi_pos = 8326
set osd_crsf_tx_pos = 8261
set osd_crsf_snr_pos = 130
set osd_crsf_rssi_pos = 8389
set osd_flymode_pos = 8362
set osd_throttle_pos = 8298
set osd_vtx_channel_pos = 2100
set osd_crosshairs_pos = 8664
set osd_ah_pos = 8409
set osd_current_pos = 86
set osd_craft_name_pos = 9239
set osd_gps_speed_pos = 9064
set osd_gps_sats_pos = 8279
set osd_home_dir_pos = 9053
set osd_home_dist_pos = 9044
set osd_altitude_pos = 9191
set osd_warnings_pos = 9108
set osd_avg_cell_voltage_pos = 9027
set osd_disarmed_pos = 9174
set osd_core_temp_pos = 999
set osd_stat_rtc_date_time = ON
set osd_stat_max_dist = ON
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set debug_mode = GYRO_SCALED
set cpu_overclock = 240MHZ
set vcd_video_system = HDZERO
set pinio_box = 40,0,0,0

# profile
profile 0

set dterm_abg_alpha = 700
set dterm_abg_boost = 750
set dterm_abg_half_life = 250
set dterm_lowpass_hz_roll = 105
set dterm_lowpass_hz_pitch = 105
set dterm_lowpass_hz_yaw = 105
set dterm_lowpass2_hz_roll = 210
set dterm_lowpass2_hz_pitch = 210
set iterm_relax_cutoff = 8
set iterm_relax_cutoff_yaw = 18
set pidsum_limit_yaw = 1000
set throttle_boost = 6
set feathered_pids = 40
set i_decay = 7
set emu_boost = 120
set emu_boost_yaw = 105
set emu_boost_limit = 52
set emu_boost_limit_yaw = 50
set emu_gravity = 75
set axis_lock_multiplier = 7
set axis_lock_hz = 8
set p_pitch = 67
set i_pitch = 120
set d_pitch = 45
set p_roll = 62
set i_roll = 110
set d_roll = 43
set p_yaw = 65
set i_yaw = 110
set d_yaw = 3
set df_yaw = 40
set linear_thrust_low_output = 50
set linear_thrust_high_output = 28
set linear_throttle = ON
set mixer_laziness = ON

# rateprofile
rateprofile 0

set thr_mid = 22
set thr_expo = 35
set rates_type = RACEFLIGHT
set roll_rc_rate = 43
set pitch_rc_rate = 43
set yaw_rc_rate = 37
set roll_expo = 35
set pitch_expo = 35
set yaw_expo = 43
set roll_srate = 102
set pitch_srate = 102
set yaw_srate = 90
set rate_center_weight = 15
set rate_end_weight = 20
set tpa_rate_p = 0
set tpa_rate_i = 0
set tpa_rate_d = 0
set tpa_breakpoint = 1330

nerdCopter, 2023-02-07:
EmuFlight was forked from ButterFlight which was forked from BetaFlight 3.5.x -- No additional GPS work has been done to this legacy GPS code.  EmuFlight has planned to rebase from a more modern Betaflight version, but work has stalled.  There will be no effort to fix this in current code, but we do accept Pull-Requests.

DiehardFPV, 2023-02-08:
> EmuFlight was forked from ButterFlight which was forked from BetaFlight 3.5.x -- No additional GPS work has been done to this legacy GPS code. EmuFlight has planned to rebase from a more modern [...]

After testing, GPS rescue works, is it possible to optimize this code?