emuflight/EmuFlight #882: Target request for BetaFPVF72235A AIO - ICM42688

DutchFPV, 2023-04-02:
* Please provide the target name. - BETAFPVF722_ICM42688
* Please provide vendor and model of the FC. - BetaFPVF72235A AIO FC/ESC- ICM42688
* Please provide a link to the vendor specifications - 
https://betafpv.com/collections/flight-controller-1/products/toothpick-f722-2-6s-aio-brushless-flight-controller-35ablheli_s

**CLI dump or Unified Target** - BetaFPV has a dedicated page for target and CLI defaults.  Link is below.
* Please provide a CLI `dump` of **DEFAULTS** or link to unified target -
https://support.betafpv.com/hc/en-us/articles/11709633442329-CLI-for-Toothpick-F722-2-6S-AIO-Brushless-Flight-Controller-35A-ICM42688

BeauBrewski, 2023-05-21:
Working on test target
https://github.com/emuflight/EmuFlight/pull/891

BeauBrewski, 2023-05-21:
Needs tested:
Analog version
[EmuFlight_0.4.2_BETAFPVF722_Build_11ab40af8.zip](https://github.com/emuflight/EmuFlight/files/11524608/EmuFlight_0.4.2_BETAFPVF722_Build_11ab40af8.zip)

HDZero Version:
[EmuFlight_0.4.2_BETAFPVF722_Build_4bd7ae4df_ICM42688P_HDZERO.zip](https://github.com/emuflight/EmuFlight/files/11524611/EmuFlight_0.4.2_BETAFPVF722_Build_4bd7ae4df_ICM42688P_HDZERO.zip)

DutchFPV, 2023-05-21:
Analog version loads and has gyro response.  Tested on an HDZ build but all features seemed good and working and the drone hovered and flew LOS 

HDZ version also good and working.  Flies as expected.  HDZ OSD has some small artifacts and misalignment of menus, but nonetheless seems fully functional.

nerdCopter, 2023-05-27:
slightly updated. would be curious if artifacts persist.
and are pidloops a factor? as snother target hit 100% cpu at 8k.

[EmuFlight_0.4.2_BETAFPVF722_Build_3c25c2936_ICM42688P_Analog.hex.zip](https://github.com/emuflight/EmuFlight/files/11582646/EmuFlight_0.4.2_BETAFPVF722_Build_3c25c2936_ICM42688P_Analog.hex.zip)
[EmuFlight_0.4.2_BETAFPVF722_Build_c1ef507ac_ICM42688P_HDZero.hex.zip](https://github.com/emuflight/EmuFlight/files/11582647/EmuFlight_0.4.2_BETAFPVF722_Build_c1ef507ac_ICM42688P_HDZero.hex.zip)

DutchFPV, 2023-05-27:
Ok so this is good.  Gyro is detected and I can get proper board orientation.  I expect we can work with this.  However I am unable to detect motors 3 and 4 with the new hex.  Neither in Motors tab or Bluejay.  Flashing back to the original hex, all 4 motors are detected fine.  I noticed that the resource assignments for motors 1-4 are all different between the old hex and the new one.  Not sure if that means that same resource assignments will work.  But if you need a CLI dump let me know.  Thanks so much for this.  We're close!

BeauBrewski, 2023-05-27:
The target with the high CPU% was an F411 board, and this has been an issue with those boards since 0.4.0, so may not be related to the new code.

DutchFPV, 2023-05-27:
Yes, meant to note that same thing.  With this board CPU% was only like 3-4.  So I can confirm its not an issue with this FC.

nerdCopter, 2023-05-28:
okay, this is confusing since nothing in the target was changed.  only changed an `#endif ` to where it was supposed to go.
https://github.com/emuflight/EmuFlight/pull/891/commits/cd6e9e3f27a800de098bc6dc19567b3e1341315d

cc: @Peck07

nerdCopter, 2023-05-28:
analog is same motor issue?

for notes: we know `11ab40a` analog worked. but merged in into a custom unofficial HD branch (4bd7ae4) in my github repo (`20230519_HDZero+master+ICM426XX`).

DutchFPV, 2023-05-28:
Yes, I'm testing analog first on this build.  So everything so far is in reference to the analog hex.

nerdCopter, 2023-05-28:
rebuilt just in case was a bad build: 
[EmuFlight_0.4.2_BETAFPVF722_Build_3c25c2936_ICM42688P_Analog_rebuild.hex.zip](https://github.com/emuflight/EmuFlight/files/11583557/EmuFlight_0.4.2_BETAFPVF722_Build_3c25c2936_ICM42688P_Analog_rebuild.hex.zip)

these are the old/working ones:
[BETAFPVF722_ICM42688P_previously_Known_good.zip](https://github.com/emuflight/EmuFlight/files/11583556/BETAFPVF722_ICM42688P_previously_Known_good.zip)

DutchFPV, 2023-05-28:
No luck.  Still fails to detect motors 3 and 4

BeauBrewski, 2023-05-28:
@nerdCopter 
From what I'm seeing in your target files on that branch:

1. In target.c, remove line 31
```
  DEF_TIM(TIM8,  CH3,  PC8, TIM_USE_ANY,               0, 0), // USE FOR CAMERA CONTROL
``` 
which assigns a timer/channel to Camera Control pin C8, as this does not exist in the BF config nor the BETAFPV cli config from above for this FC.  (You'll also want to remove line 36 `#define CAMERA_CONTROL_PIN                               PC8` in target.h file as well, as this resource does not exist)

2. After removing the above, you'll then want to adjust the Timer Count and assignment in target.h to be as follows:
```
// timers
#define USABLE_TIMER_CHANNEL_COUNT                      7
#define USED_TIMERS ( TIM_N(1) |TIM_N(2) | TIM_N(3) | TIM_N(4) )
```
3. Lastly, this may not be necessary, but since it is for an AIO board, might want to try adding `#define USE_SERIAL_4WAY_BLHELI_INTERFACE` to the target.h file.

nerdCopter, 2023-05-28:
thanks @BeauBrewski  :muscle:  , weird that it worked before. maybe the broken code stopped before that define :man_shrugging: ...  i'll do that today.

BeauBrewski, 2023-05-28:
Yeah, I was looking at old targets before this rebase and thats how its always been, so now I'm a little stumped. Worth a try, however.

nerdCopter, 2023-05-28:
@DutchFPV , please try both of these: 
[betafpvf722_icm42688p_Analog_BeauRecommends.zip](https://github.com/emuflight/EmuFlight/files/11584906/betafpvf722_icm42688p_Analog_BeauRecommends.zip)

DutchFPV, 2023-05-28:
Both work fine in Emuflight but neither is giving me motors 3 or 4 in Emu configurator or Blujay config.

DutchFPV, 2023-05-28:
OMG apologies, I got my threads crossed.  Jumped here from the JECMCU target thread, and gave you the wrong info.  To be clear, this target is fine so far.  I was doing some parts swapping, but plan to test the HDZ OSD this afternoon.  The motor issue is related to the other target.  I'll update there.  You can disregard my last.

nerdCopter, 2023-05-28:
great news!

DutchFPV, 2023-05-28:
Apologies for confusing this target with the other one.  All functions are fine.  I will update you on HDZ OSD shortly

DutchFPV, 2023-05-29:
OK following up now that I'm back on the correct target.  As reported previously gyro, arm, etc are fine and working.  CPU utilization is very low and not an issue.  HDZero artifacts and misalignments still remain unchanged from the prior hex best I can tell.

nerdCopter, 2023-06-02:
I wouldn't mind these hex tested before merging to master:
[EmuFlight_0.4.2_BETAFPVF722_Build_7e2777b2b_ICM42688P_Analog_Final.hex.zip](https://github.com/emuflight/EmuFlight/files/11637106/EmuFlight_0.4.2_BETAFPVF722_Build_7e2777b2b_ICM42688P_Analog_Final.hex.zip)
[EmuFlight_0.4.2_BETAFPVF722_Build_488c1c625_ICM42688P_HDZero.hex.zip](https://github.com/emuflight/EmuFlight/files/11637105/EmuFlight_0.4.2_BETAFPVF722_Build_488c1c625_ICM42688P_HDZero.hex.zip)

PR #905 (2023-08-23): [Target] BETAFPVF722 add ICM42688P; fix Timers
* BETAFPVF722 add ICM42688P
* fix wrong timers https://github.com/emuflight/EmuFlight/pull/905/files
* this was flown by @DutchFPV

Closes #882
closed 2023-08-23 by nerdCopter