emuflight/EmuFlight #892: DIAT-MAMABAF405_2022B

DutchFPV, 2023-05-19:
**Vendor/Model**
* Please provide the target name:  DIAT-MAMBAF405_2022B
* Please provide vendor and model of the FC:  Diatone `MAMBAF405 MK IV 2022B (ICM42688)`
* Please provide a link to the vendor specifications.

https://www.getfpv.com/diatone-mamba-mk4-f405-mini-flight-controller-icm42688p.html

**CLI dump or Unified Target**

https://github.com/betaflight/unified-targets/blob/master/configs/default/DIAT-MAMBAF405_2022B.config

* Please provide a CLI `dump` of **DEFAULTS** or link to unified target.
  Find the unified targets here: https://github.com/betaflight/unified-targets/tree/master/configs/default

This is for a teammate who has the FC and is ready to test and feedback asap.  Thanks!

BeauBrewski, 2023-05-21:
Working on test target
https://github.com/emuflight/EmuFlight/pull/891

nerdCopter, 2023-05-27:
i have a feeling that target  `DIAT_MAMBAF405_2022B` will not work, i've learned more about target making, but much more to learn. would still like feedback to know the status.

nerdCopter, 2023-06-01:
#908

DutchFPV, 2023-06-01:
Will do.  This is for a teammate but he's eager to get it and test it.  I'll report back with BB asap.  Thanks!

DutchFPV, 2023-06-05:
Got an update this morning that the target is still not showing active gyro in the configurator.  Let me know if I can gather any further details for you.

BeauBrewski, 2023-06-30:
@DutchFPV 

With BF installed, in CLI can your friend type `status` and copy/paste results back.

Other thing that would be useful is with that last hex, to send a copy of the `dump all` so we can compare the hex results to the BF config.

nerdCopter, 2023-07-17:
@DutchFPV 
i leveled up personal knowledge, let's test this 
[EmuFlight_0.4.2_DIAT_MAMBAF405_2022B_Build_20eb01f53_Analog_needs_testing_and_feedback.hex.zip](https://github.com/emuflight/EmuFlight/files/12070467/EmuFlight_0.4.2_DIAT_MAMBAF405_2022B_Build_20eb01f53_Analog_needs_testing_and_feedback.hex.zip)

DutchFPV, 2023-07-17:
Is that an upgraded/updated hex?  I'll see if I can get you BB from my buddy.  Sorry we had both been traveling, but should have time to do this soon.  I'll update asap

nerdCopter, 2023-07-17:
> Is that an upgraded/updated hex? 

yes, new from scratch. i've learned a bit more since first attempts.

DutchFPV, 2023-07-19:
The official word is it "flew beautifully".  Here's the bb he was able to capture...
[mamba_f4_ICM.zip](https://github.com/emuflight/EmuFlight/files/12100529/mamba_f4_ICM.zip)

nerdCopter, 2023-07-19:
> The official word is it "flew beautifully". Here's the bb he was able to capture... [mamba_f4_ICM.zip](https://github.com/emuflight/EmuFlight/files/12100529/mamba_f4_ICM.zip)

Thank you both!! discord for any further needs.

PR #908 (2023-08-24): [Target] DIAT_MAMBAF405_2022B - MPU6000, MPU6500, ICM42688P
* `DIAT_MAMBAF405_2022B` MPU6000, ICM42688P
* tested as flying with blackbox
* no baro dps310 support

Closes #892
closed 2023-08-24 by nerdCopter