emuflight/EmuFlight #894: TARGET REQUEST - JHEMCU GHF722AIO - ICM - ESC40A

DutchFPV, 2023-05-21:
* Please provide the target name.  - JHEF7Dual
* Please provide vendor and model of the FC. - JHEMCU  GHF722AIO - ICM - ESC-40A
* Please provide a link to the vendor specifications.
https://www.aliexpress.com/i/3256804754098306.html?gatewayAdapt=4itemAdapt

**CLI dump or Unified Target**
https://github.com/betaflight/unified-targets/blob/master/configs/default/JHEF-JHEF7DUAL.config

One more ICM FC.  I have one of these on hand for an upcoming build.  Just wanted to get it on the list.

nerdCopter, 2023-05-24:
needs thorough testing with focus on safety

please submit blackbox with 
```
set debug_mode = GYRO_SCALED
save
```

please feedback
```
Gyro
Accelerometer
Motors
Beeper
Onboard ADC
OSD (analog)
Smart Audio
Barometer
Blackbox
LED
GPS
```

[EmuFlight_0.4.2_JHEF_JHEF7DUAL_Build_b23ef6239_ICM42688P_Analog.hex.zip](https://github.com/emuflight/EmuFlight/files/11559330/EmuFlight_0.4.2_JHEF_JHEF7DUAL_Build_b23ef6239_ICM42688P_Analog.hex.zip)

@DutchFPV , if this one test-hovers i can make HD for deep-testing.

DutchFPV, 2023-05-25:
Not getting any gyro in the configurator setup.  FYI this is the AIO board that only has one gyro.  I do see the 'gyro_to_use' option in the CLI and tried both to no avail.

nerdCopter, 2023-05-25:
Mulligan
[EmuFlight_0.4.2_JHEF_JHEF7DUAL_Build_80159eda9_ICM42688P_Analog.hex.zip](https://github.com/emuflight/EmuFlight/files/11567311/EmuFlight_0.4.2_JHEF_JHEF7DUAL_Build_80159eda9_ICM42688P_Analog.hex.zip)

DutchFPV, 2023-05-26:
Seems to have no USB connection after flashing.  Flashed back to the old target and it connected fine

nerdCopter, 2023-05-26:
i'll keep trying...
[EmuFlight_0.4.2_JHEF_JHEF7DUAL_Build_1f2e225ae_ICM42688P_Analog.hex.zip](https://github.com/emuflight/EmuFlight/files/11578192/EmuFlight_0.4.2_JHEF_JHEF7DUAL_Build_1f2e225ae_ICM42688P_Analog.hex.zip)

DutchFPV, 2023-05-26:
Much appreciated, but still no USB.  Status led never comes on after flashing and no connection is made to PC.

nerdCopter, 2023-05-26:
[EmuFlight_0.4.2_JHEF_JHEF7DUAL_Build_8c7ac7321_ICM42688P_Analog.hex.zip](https://github.com/emuflight/EmuFlight/files/11579818/EmuFlight_0.4.2_JHEF_JHEF7DUAL_Build_8c7ac7321_ICM42688P_Analog.hex.zip)

DutchFPV, 2023-05-26:
It breaks my heart to tell you this, but still no USB.  As soon as I flash back to the original JHEF7DUAL target it connects fine again.

nerdCopter, 2023-05-27:
@DutchFPV , thanks for patience. Instead of creating target from scratch, I'll try to add icm42688p to the existing one. I already see that serial count is 7 not 6 like I've defined.

It may be Saturday evening or Sunday. Unsure of my schedule.

nerdCopter, 2023-05-27:
i updated the existing target. it should work :crossed_fingers: 
[EmuFlight_0.4.2_JHEF7DUAL_Build_3c25c2936_ICM42688P_Analog.hex.zip](https://github.com/emuflight/EmuFlight/files/11582559/EmuFlight_0.4.2_JHEF7DUAL_Build_3c25c2936_ICM42688P_Analog.hex.zip)
[EmuFlight_0.4.2_JHEF7DUAL_Build_c1ef507ac_ICM42688P_HDZero.hex.zip](https://github.com/emuflight/EmuFlight/files/11582558/EmuFlight_0.4.2_JHEF7DUAL_Build_c1ef507ac_ICM42688P_HDZero.hex.zip)

However, i'm including the from-scratch as well, because i'm attempting to fully learn/grasp target-creation: (This is the one that failed multiple times above.)
[EmuFlight_0.4.2_JHEF_JHEF7DUAL_Build_3c25c2936_ICM42688P_Analog_untested.hex.zip](https://github.com/emuflight/EmuFlight/files/11582563/EmuFlight_0.4.2_JHEF_JHEF7DUAL_Build_3c25c2936_ICM42688P_Analog_untested.hex.zip)
[EmuFlight_0.4.2_JHEF_JHEF7DUAL_Build_c1ef507ac_ICM42688P_HDZero_untested.hex.zip](https://github.com/emuflight/EmuFlight/files/11582567/EmuFlight_0.4.2_JHEF_JHEF7DUAL_Build_c1ef507ac_ICM42688P_HDZero_untested.hex.zip)

BeauBrewski, 2023-05-28:
@nerdCopter Might be nothing, but remember in the past when we had issues with USB detect on longer target names (and yet it was difficult to reproduce with any reliability)? Might want to leave off the `JHEF_` at the beginning of the target name and see if that helps.

Also when doing Serial Count, it includes `USE_VCP` + `USE_UART#'s` + any `USE_SOFTSERIAL#'s`
For the `JHEF7DUAL` target, this will be USB+6 uarts, so serial count =7. 
If you were doing serial count 6, that might also be an issue, but I would have thought it would cut off UART6 and not USB 🤷‍♂️

nerdCopter, 2023-05-28:
i figured out the total + vcp yesterday. but never thought about the long name. thanks!  but my from-scratch definitely has some differences than original.  i think the adding icm42688p to the original will work, just needs testing.  fwiw, the long-name hex above (yesterday) does have a `7` count.

DutchFPV, 2023-05-28:
Yes, def unplugging and replugging.  I don't know if it makes a difference either, but I always do it that way.

BeauBrewski, 2023-05-28:
@DutchFPV I figured you were, but gotta ask just in case 😄 

@nerdCopter 
Bro, I think I figured it out, your target.mk file is defined for a F745 flight controller, not F722 😸 
Change line 1 in target.mk to:
`F7X2RE_TARGETS += $(TARGET)`

nerdCopter, 2023-05-28:
off-subject, longest target names:
[image]

nerdCopter, 2023-05-28:
i'll await feedback on 
```
EmuFlight_0.4.2_JHEF7DUAL_Build_3c25c2936_ICM42688P_Analog.hex.zip
EmuFlight_0.4.2_JHEF7DUAL_Build_c1ef507ac_ICM42688P_HDZero.hex.zip
```
before uploading any more

DutchFPV, 2023-05-28:
Guys, OMG, I got my threads crossed on the 2 active target requests.  The build numbers were the same and I got them crossed up working on both in parallel.  This target is working fine.  No ESC issue. I'll report on flight as soon as it starts raining

nerdCopter, 2023-05-28:
blackbox flight if you can

closed 2023-10-31 by nerdCopter