emuflight/EmuFlight #897: [Target Request] TMOTORF7 (BMI270)

BeauBrewski, 2023-05-29:
**Vendor/Model**
TMOTOR F722 with BMI270
`TMOTORF7`

**CLI dump or Unified Target**
based off existing target with BMI270 defines and drivers added.

BeauBrewski, 2023-05-29:
NEEDS TESTED! *PROPS OFF FOR SAFETY!*

[EmuFlight_0.4.2_TMTR_TMOTORF7_Build_719010445.zip](https://github.com/emuflight/EmuFlight/files/11587494/EmuFlight_0.4.2_TMTR_TMOTORF7_Build_719010445.zip)

Check motor temps when flying!

NOTE When setting up in configurator for bmi270, the gyro/pidloop will report wrong. MUST SET to 8k which just means denominator = 1. when denominator is 1, this is gyro default for bmi270, which is about 3.2k.
or in CLI:

set gyro_sync_denom = 1
set pid_process_denom = 1
save

nerdCopter, 2023-06-07:
updated.  still need feedback. no need to set denoms manually.[EmuFlight_0.4.2_TMTR_TMOTORF7_Build_b2347716b_BMI270_Analog_needs_flight_testing.hex.zip](https://github.com/emuflight/EmuFlight/files/11679451/EmuFlight_0.4.2_TMTR_TMOTORF7_Build_b2347716b_BMI270_Analog_needs_flight_testing.hex.zip)

agorskiy12, 2023-07-30:
tried it and flew really well

nerdCopter, 2023-07-31:
> tried it and flew really well

thank you!!
welcome to feedback on these items if you can:
[Yes/No/Untested] hex Flash & subsequent USB connect
[Yes/No/Untested] Gyro 
[Yes/No/Untested] Accelerometer
[Yes/No/Untested] Motors
[Yes/No/Untested] Onboard ADC
[Yes/No/Untested] OSD (analog)
[Yes/No/Untested] Smart Audio controls
[Yes/No/Untested] Beeper/Alarm switch
[Yes/No/Untested] Barometer (not applicable unless purchased with Baro)
[Yes/No/Untested] Blackbox  flash-chip
[Yes/No/Untested] LED Strip
[Yes/No/Untested] GPS

I have this FC as well, but due to VTX problems cannot fully test.
what is your ADC Battery `Scale` & `Divider` set to? i'm using 108 & 110, but unsure if this is accurate.

nerdCopter, 2023-11-02:
https://github.com/emuflight/EmuFlight/pull/944

closed 2023-11-06 by nerdCopter