emuflight/EmuFlight #898: [Target Request] BETAFPVF411 (BMI270)

BeauBrewski, 2023-05-29:
**Vendor/Model**
BETAFPVF411 w/ BMI270 (note-not the RX version)

**CLI dump or Unified Target**
https://github.com/betaflight/unified-targets/blob/master/configs/default/BEFH-BETAFPVF411.config
based off existing target with just new BMI270 drivers added.

BeauBrewski, 2023-05-29:
NEEDS TESTED! PROPS OFF FOR SAFETY!

[EmuFlight_0.4.2_BETAFPVF411_Build_719010445.zip](https://github.com/emuflight/EmuFlight/files/11587589/EmuFlight_0.4.2_BETAFPVF411_Build_719010445.zip)

Check motor temps when flying!

NOTE When setting up in configurator for bmi270, the gyro/pidloop will report wrong. MUST SET to 8k which just means denominator = 1. when denominator is 1, this is gyro default for bmi270, which is about 3.2k.
or in CLI:

set gyro_sync_denom = 1
set pid_process_denom = 1
save

whooptrooper, 2023-10-09:
Works as far as the gyro is now recognized. But it hovers with an enormous oscillation. Any further ideas how to get this to work? have a BetaFPV Meteor65 Pro with 22kkV 0802 Motors.

nerdCopter, 2023-10-09:
i would raise filters. 

<details><summary>do not over filter.</summary>

 for example, either use gyro LPF at 150ish and maybe 1 dynamic_filter notch, or dont use dynamic filter at all. or another option, dynamic_filter with 2 or 3 notches, and a very high gyro LPF or none at all.  -- to be tested of course.
</details>

for whoops, generally.
gyro cutoff maybe 150hz
dterm cutoff maybe 200hz.
roll/pitch normally about the same PIDs for each.
but for BMI, maybe not too much D on roll. bmi still being tested.

blackbox would certainly help, but 2021 Meteor65Pro did not have Blackbox. does the new versions have it?

I'd like to build a news hex however! The May 2023 is outdated. 
here: [EmuFlight_0.4.2_BETAFPVF411_Build_dc51cc9dd_BMI270_FIFO_OSR4_Analog.hex.zip](https://github.com/emuflight/EmuFlight/files/12850597/EmuFlight_0.4.2_BETAFPVF411_Build_dc51cc9dd_BMI270_FIFO_OSR4_Analog.hex.zip)
This build should be better. maybe even test it with your current configuration.

nerdCopter, 2023-10-09:
send blackbox logs if applicable, preferably via Discord.

whooptrooper, 2023-10-10:
@nerdCopter Thx! I'm gonna try it tomorrow and tell you. Can I share the log here in the issue?

whooptrooper, 2023-10-10:
@nerdcoopter It works way better with the new build! I had just a little left oscillations by default and changed the lowpass cutoff to 150Hz as you said. I think the oscillations are gone now, but I would really appreciate you to have a look at my log. I did a little logging flight with angle mode hope it does the job. 
[whooptrooper Meteor 65 Pro 2022.zip](https://github.com/emuflight/EmuFlight/files/12861342/whooptrooper.Meteor.65.Pro.2022.zip)

nerdCopter, 2023-10-10:
need
```
set debug_mode = gyro_scaled
save
```
and can you set rates as betaflight rates for blackbox? our Eme blackbox exploere does not render setpoint correctly for kiss nor actual rates. :(

whooptrooper, 2023-10-11:
Oh yeah sorry about that. I will do it later!

nerdCopter, 2023-10-11:
[image]

need a lot more D (roll/pitch), like a lot.  you'll have to raise them and check via PTB's step-response tool.
[image]

need higher yaw P, 75 to 90+ to your liking.  50's are fine for betaflight, but Emu pid controller different.
you can probably go yaw D 0.  0-5max  where 0 for P of 70's and 5 for P of 90's+

hard to say on dterm filters, but i'm thinking
 Dterm LPF1 type PT1 at 150 to 200, and 
 Dterm LPF2 type PT1 at 350 to 400
 
Roll a little noisy, as i would suspect on BMI270
[image]
compared to pitch
[image]

nerdCopter, 2023-10-11:
red line is the "goal"; not always achievable.
[image] (copyright)Brian White/PTB

nerdCopter, 2023-11-08:
resolved via https://github.com/emuflight/EmuFlight/pull/937

closed 2023-11-08 by nerdCopter