emuflight/EmuFlight #901: [Target Request] SPEEDYBEEF405V2 (BMI270)

BeauBrewski, 2023-05-29:
**Vendor/Model**
`SPEEDYBEEF405V2` (pilot asked for v2, only seeing v3 at moment, will clarify and get a dump if its v2)
https://www.speedybee.com/speedybee-f405-v3-bls-50a-30x30-fc-esc-stack/

**CLI dump or Unified Target**
https://github.com/betaflight/unified-targets/blob/master/configs/default/SPBE-SPEEDYBEEF405V3.config

BeauBrewski, 2023-06-03:
NEEDS TESTED!!!
*PROPS OFF FOR SAFETY!* 
Hover and check motor temps often!

[EmuFlight_0.4.2_SPEEDYBEEF405V3_Build_f0fb68932.zip](https://github.com/emuflight/EmuFlight/files/11642702/EmuFlight_0.4.2_SPEEDYBEEF405V3_Build_f0fb68932.zip)

NOTE When setting up in configurator for bmi270, the gyro/pidloop will report wrong. MUST SET to 8k which just means denominator = 1. when denominator is 1, this is gyro default for bmi270, which is about 3.2k.
or in CLI:
```
set gyro_sync_denom = 1
set pid_process_denom = 1
save
```
✔️/❌ Flash/SD Card
✔️/❌ Gyro
✔️/❌ Accelerometer
✔️/❌ Motors
✔️/❌ Onboard ADC
✔️/❌ OSD (analog)
✔️/❌ Smart Audio controls (OSD and CLI)
✔️/❌ Alarm switch
✔️/❌ Barometer
✔️/❌ Blackbox
✔️/❌ LED
✔️/❌ GPS
Or any other features, issues you noticed?

BeauBrewski, 2023-06-04:
Per FB Pilot:
"It flies great, no hot motors, nothing strange. put about 3 packs thru already...and the speedybee app works fine as well.
Just the default baud rate for uart4 (the Bluetooth) is just incorrect minor detail just had to set it the same as the others so simple fix."

Will update the baud rate in config.c file. 🚀

BeauBrewski, 2023-06-07:
Here's a test target for the HDZero/Walksnail dev-unstable version. Must install dev-unstable configurator as well:
https://github.com/emuflight/dev-unstable/releases/tag/cfg-598

[EmuFlight_0.4.2_SPEEDYBEEF405V3_Build_728172592_BMI270_HDZero_needs_testing.zip](https://github.com/emuflight/EmuFlight/files/11679501/EmuFlight_0.4.2_SPEEDYBEEF405V3_Build_728172592_BMI270_HDZero_needs_testing.zip)

nerdCopter, 2023-09-07:
Another build. needs testing. created via script + manual modification.

[EmuFlight_0.4.2_SPBE_SPEEDYBEEF405V3_BMI270_needs_testing.zip](https://github.com/emuflight/EmuFlight/files/12553173/EmuFlight_0.4.2_SPBE_SPEEDYBEEF405V3_BMI270_needs_testing.zip)

note: bmi270 will show 8k/8k, ignore this, it is 3.2k/3.2k in hardware
note: HDZero requires dev-unstable Configurator: https://github.com/emuflight/dev-unstable/releases/tag/cfg-598

No guarantees -- Needs thorough testing and feedback.
Test gyro/accel on USB (Configurator sensors tab) first.
Test orientation.
Then test props off.
Then test-hover.
check motor temperatures, do not full-send hot motors.
please test/feedback any/all features applicable:

[Yes/No/Untested] hex Flash & subsequent USB connect
[Yes/No/Untested] Gyro 
[Yes/No/Untested] Accelerometer
[Yes/No/Untested] Motors
[Yes/No/Untested] Onboard ADC
[Yes/No/Untested] OSD (analog)
[Yes/No/Untested] Smart Audio controls
[Yes/No/Untested] Beeper/Alarm switch
[Yes/No/Untested] Barometer
[Yes/No/Untested] Blackbox  flash-chip/sdcard
[Yes/No/Untested] LED Strip
[Yes/No/Untested] GPS

nerdCopter, 2023-11-08:
> Per FB Pilot: "It flies great, no hot motors, nothing strange. put about 3 packs thru already...and the speedybee app works fine as well. Just the default baud rate for uart4 (the Bluetooth) is just [...]

@BeauBrewski , can you send me the config.c file for the baud? TY!

nerdCopter, 2024-04-18:
resolved by https://github.com/emuflight/EmuFlight/pull/955. released to 0.4.2 assets.
did not check/fix baud.

closed 2024-04-18 by nerdCopter