emuflight/EmuFlight #909: [target request] RUSHBLADEF7HD ICM42688P

nerdCopter, 2023-06-04:
**Rush Blade F7 HD ICM42688P**
* `RUSHBLADEF7HD` aka `RUSH_BLADE_F7_HD` 
* https://rushfpv.net/products/rushfpv-rush-blade-f722-30x30-flight-controller-for-dji-digital-fpv-system-v2-version
* https://github.com/betaflight/unified-targets/blob/master/configs/default/RUSH-BLADE_F7_HD.config
* https://github.com/betaflight/config/blob/master/configs/BLADE_F7_HD/config.h

nerdCopter, 2023-06-04:
[EmuFlight_0.4.2_RUSHBLADEF7HD_Build_17e3b27d8_ICM42688P_MPU6000_Analog.hex.zip](https://github.com/emuflight/EmuFlight/files/11646190/EmuFlight_0.4.2_RUSHBLADEF7HD_Build_17e3b27d8_ICM42688P_MPU6000_Analog.hex.zip)

Needs thorough testing and feedback.
Test gyro/accel on USB (Configurator sensors tab) first.
Test orientation.
Then test props off.
Then test-hover.
check motor temperatures, do not full-send hot motors.
please test/feedback any/all features applicable:
✔️/❌ Flash
✔️/❌ Gyro
✔️/❌ Accelerometer
✔️/❌ Motors
✔️/❌ Onboard ADC
✔️/❌ OSD (analog)
✔️/❌ Smart Audio controls (OSD and CLI)
✔️/❌ Alarm switch
✔️/❌ Barometer
✔️/❌ Blackbox
✔️/❌ BlueTooth (if applicable)
✔️/❌ LED
✔️/❌ GPS

RogerGrafix, 2023-08-19:
Target flashed, needed to close and open configurator to be able to connect with fc. Gyro working fine on usb, arrow directions ok.

RogerGrafix, 2023-08-20:
Ok so, hoover test done and 2 packs after. All so far good. No issues so far. I keep testing it.