emuflight/EmuFlight #912: [target request] CRAZYBEEF4FR BMI270

nerdCopter, 2023-06-06:
**mobula6 CRAZYBEEF4FR **
* `CRAZYBEEF4FR` and maybe non-FR variants
* vendor specifications.
* https://github.com/betaflight/config/blob/master/configs/CRAZYBEEF4FR/config.h
* https://github.com/betaflight/unified-targets/blob/master/configs/default/HAMO-CRAZYBEEF4FR.config

requested by @ VH

xxxvik-xakerxxx, 2023-06-07:
Tested only my CRAZYBEEF4FR ver board 4.2
fix vtx problem (signal lost)- use this https://github.com/emuflight/EmuFlight/issues/796

[EmuFlight_0.4.2_CRAZYBEEF4FR_Build_uncommitted_20230607EEST.hex.zip](https://github.com/emuflight/EmuFlight/files/11671930/EmuFlight_0.4.2_CRAZYBEEF4FR_Build_uncommitted_20230607EEST.hex.zip)

not tested use make test

my local error

`linking ../../obj/test/alignsensor_unittest/alignsensor_unittest 
ld: library not found for -lBlocksRuntime
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [../../obj/test/alignsensor_unittest/alignsensor_unittest] Error 1
make: *** [test] Error 2`

xxxvik-xakerxxx, 2023-06-07:
Dump my config 
<details><summary>Details</summary>
<p>

```
# version
# EmuFlight / CRAZYBEEF4FR (C4FR) 0.4.2 Jun  7 2023 / 00:34:29 (uncommitted_20230607EEST) MSP API: 1.51

board_name 
manufacturer_id 

# name
name mobula

# resources
resource OSD 1 C15
resource MOTOR 1 B10
resource MOTOR 2 B06
resource MOTOR 3 B07
resource MOTOR 4 B08
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 A02
resource PWM 2 A09
resource PWM 3 A10
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource TRANSPONDER 1 A00
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 NONE
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource LED 1 C13
resource LED 2 NONE
resource LED 3 NONE
resource INVERTER 1 NONE
resource ESCSERIAL 1 B02
resource SPI_SCK 1 A05
[... 734 lines cut ...]
set yaw_expo = 0
set roll_srate = 0
set pitch_srate = 0
set yaw_srate = 0
set das_yaw_with_roll_input = 0
set das_roll_with_yaw_input = 0
set roll_pitch_mag_expo = 0
set rate_center_sensitivity = 100
set rate_end_sensitivity = 100
set rate_center_correction = 10
set rate_end_correction = 10
set rate_center_weight = 0
set rate_end_weight = 0
set tpa_rate_p = 100
set tpa_rate_i = 165
set tpa_rate_d = 95
set tpa_breakpoint = 1400
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set vbat_comp_type = OFF
set vbat_comp_ref = 37
```

</p>
</details>

nerdCopter, 2023-06-07:
our make test has not been working for awhile; avoid it

nerdCopter, 2023-11-06:
resolved with https://github.com/emuflight/EmuFlight/pull/916

closed 2023-11-06 by nerdCopter