emuflight/EmuFlight #924: [target] JHEFF411 add ICM42688P

torju01, 2023-07-18:
Hi Beau :-) 
please to ask you and read you again.
here is the "bad AIO boy" !
FC desciption: JHEMCU Ruibet F411 AIO 12A   (first realise, with baro and blackbox)
( https://www.jhemcu.com/e_productshow/?72-JHEMCU-Ruibet-F4AIO-12A-72.html )
Problem: no gyro movement with actuel JHEFF411 emuflight target
perhaps target need the ICM42688P gyro !?

Last JHEFF411 betaflight target work properly.
( https://github.com/betaflight/unified-targets/blob/master/configs/default/JHEF-JHEF411.config )

Thank you very much Beau for your help
Best regard
Seb

nerdCopter, 2023-07-18:
[EmuFlight_0.4.2_JHEF411_Build_94ddc55b5_ICM42688P_Analog.hex.zip](https://github.com/emuflight/EmuFlight/files/12083278/EmuFlight_0.4.2_JHEF411_Build_94ddc55b5_ICM42688P_Analog.hex.zip)
please report findings.

torju01, 2023-07-19:
Hello NerdCopter,
super, gyro is back :-) and it fly great !
thank you very much.   
"You know what ? i am happy" :-) 
Droopy

PR #925 (2023-08-23): [Target] JHEF411 -  add ICM42688P
* Gyro works: https://github.com/emuflight/EmuFlight/issues/924#issuecomment-1641857449
* closes #924