emuflight/EmuFlight #926: HAKRC F722D (not AIO) uses MATEKF722SE firmware in betaflight, but EmuFlight can NOT recognize gyro...

kaji331, 2023-08-06:
Unified Target
https://github.com/betaflight/unified-targets/blob/master/configs/default/MTKS-MATEKF722SE.config

It flies just so so in BetaFlight (I don't like it's filter for this MPU6500 board). I tried HAKRC F722D firmware and MATEKF722SE firmware in EmuFlight 0.4.2, neither can recognize the gyro.

meoso, 2023-08-07:
what gyro, we only recently merged ICM42688P support.  BMI270 is still in development and un-merged, but i can probably build it for you.

mpu6000 and 6500 already supported, AFAIK.  that link does not report newer gyros.

nerdCopter, 2023-08-07:
what gyro, we only recently merged ICM42688P support. BMI270 is still in development and un-merged, but i can probably build it for you.

mpu6000 and 6500 already supported, AFAIK. that link does not report newer gyros.

kaji331, 2023-08-08:
> what gyro, we only recently merged ICM42688P support. BMI270 is still in development and un-merged, but i can probably build it for you.
> 
> mpu6000 and 6500 already supported, AFAIK. that link does not report newer gyros.
> 
> Additionally, HD is still in development as well, so i would need to know if you need Analog or HD.

The gyro should be mpu6500 (65 on the mpu chip) and I uses analog. I don't know why betaflight can fly but emuflight can not, I really like emu's filter.

nerdCopter, 2023-08-08:
@kaji331 , in betaflight, please paste the output of cli command `status`.

Also, i just reviewed our `MATEKF722SE` target definition, it seems it is configured statically for mpu6000 on gyro 1, and mpu6500 on gyro 2.    if there is 6500 as hardware gyro 1, this is the reason.

if you can validate hardware and positions, i can modify some test code and submit here for testing.

kaji331, 2023-08-09:
> @kaji331 , in betaflight, please paste the output of cli command `status`.
> 
> ~Also, i just reviewed our `MATEKF722SE` target definition, it seems it is configured statically for mpu6000 on gyro 1, or mpu6500 on gyro 2. if there is 6500 as hardware gyro 1, this is the reason it would fail.~ [edit: this statement likely incorrect.]
> 
> ~if you can validate hardware and positions,~ i can modify some test code and submit here for testing.

# status
MCU F722 Clock=216MHz, Vref=3.26V, Core temp=43degC
Stack size: 2048, Stack address: 0x20010000
Configuration: CONFIGURED, size: 4034, max available: 16384
Devices detected: SPI:1, I2C:1
Gyros detected: gyro 2 locked dma
GYRO=MPU6500, ACC=MPU6500, BARO=DPS310
OSD: MAX7456
System Uptime: 36 seconds, Current Time: 2023-08-09T00:48:52.119+00:00
CPU:51%, cycle time: 124, GYRO rate: 8064, RX rate: 15, System rate: 9
Voltage: 37 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 1
SD card: Startup failed
Arming disable flags: RXLOSS CLI MSP RPMFILTER

#

kaji331, 2023-08-09:
> maybe `set gyro_to_use = SECOND`, `save`.

I tried this, but no use...

nerdCopter, 2023-08-09:
please test [EmuFlight_0.4.2_MATEKF722SE_Build_2838323eb.hex.zip](https://github.com/emuflight/EmuFlight/files/12302996/EmuFlight_0.4.2_MATEKF722SE_Build_2838323eb.hex.zip)

kaji331, 2023-08-09:
Hi, I tried MATEKF722SE 0.4.1 again. This time, I found after I chose second gyro in cli, it can find gyro! If I save some parameters in configurator GUI and restart, it can't find gyro again. I have to connect fc again and enter 'save' in cli, then it find the gyro again after restart...but the BMP280 baro is still not found anyway.

nerdCopter, 2023-08-09:
DPS310 is not supported in EmuFlight yet.
I do not understand the gyro issue :( -- maybe still try the hex i uploaded just in case.

kaji331, 2023-08-10:
> please test [EmuFlight_0.4.2_MATEKF722SE_Build_2838323eb.hex.zip](https://github.com/emuflight/EmuFlight/files/12302996/EmuFlight_0.4.2_MATEKF722SE_Build_2838323eb.hex.zip)

Now, I tried your uploaded firmware 0.4.2, this time I reconnected fc some times and save parameters in gui or cli. So far the gyro can always be found!:+1

nerdCopter, 2023-08-10:
interesting comment here: https://github.com/betaflight/betaflight/pull/13013#issuecomment-1666884302
Betaflight, matek, dual gyro, but icm40206 also reverts to gyro 1 after first save.  wondering if this is an ancient bug and/or matek specific.