emuflight/EmuFlight #945: GEPRC F411 35A AIO softserial (T11 R11 pin)

Alexx2321979, 2023-11-02:
**Describe the bug**
A clear and concise description of what the bug is.
pin T11 R11 not work softserial gps
**To Reproduce**
Steps to reproduce the behavior:
what i need doo this trabl?
**Expected behavior**
A clear and concise description of what you expected to happen.
softserial this pin work whis gps module
**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.
GEPRC F411 35A AIO in drone geprc rocket lite.
**Additional context**
Add any other context about the problem here.
please help

nerdCopter, 2023-11-02:
EmuFlight is based from ButterFlight/Betaflight 3.5.x . EmuF GPS support is severely lacking and I do not know much about GPS configuration.

GEPRCF411_AIO definition seems to already support softserial: https://github.com/emuflight/EmuFlight/blob/731894ae28f701601a044effc1cd3bdd314e2da3/src/main/target/GEPRCF411_AIO/target.h#L87-L105
I'm wondering if UART1 and SOFTSERIAL1 cannot be used at same time due to identical PINs PA0 & PB10. I do not know enough about it.
Only these Baro/Mag are supported if it's any consolation: https://github.com/emuflight/EmuFlight/blob/731894ae28f701601a044effc1cd3bdd314e2da3/src/main/target/GEPRCF411_AIO/target.mk#L9-L13

Alexx2321979, 2023-11-02:
[image]
[image]

this stock firmware. R11 T11 is softserial? i konect GPS bn180, but not work.

nerdCopter, 2023-11-02:
i would agree that 11 is softSerial. In Betaflight if you flashed 4.4 or later, you must enable GPS in cloud-build.
BetaFlight discord: https://discord.betaflight.com/
EmuFlight discord: https://discord.gg/BWqgBg3

Alexx2321979, 2023-11-02:
> you must enable GPS in cloud-build.

sorry, please give more info for this.
i have Betaflight 4.4.2 nevest, gps enable in konfiguration - UBLOX potr auto, but not work....

nerdCopter, 2023-11-02:
if you have an M10 GPS, betaflight 4.5 has support for M10.
Please go to appropriate Discord server for help and support.

closed 2023-11-05 by nerdCopter

Alexx2321979, 2023-11-08:
411 chip not work normal in softserial! 
Sbus work through uart1 and conflicts with another device on uart 1. softserial also conflicts with a device on uart 2. the output is simple 411 complete......
tested on both geprc f411aio and jhemcu f411 noxe v3
It is impossible to install 3 devices (vtx hd, cross, gps) only 2 devices. It's better to buy a board with a 405 chip

nerdCopter, 2023-11-08:
Thank you for the information.
F411 known to be less than ideal, but i did not know these details.

F411 also sometimes needs pid_denom = 2 to process all the filters and features. (https://github.com/emuflight/EmuFlight/issues/915)