emuflight/EmuFlight #999: [Target Request] CRAZYBEEF4FS (BMI270)

alber70g, 2024-04-06:
**Vendor/Model**
* Please provide the target name.  
`CRAZYBEEF4FS`

* Please provide vendor and model of the FC.  
`Happymodel CrazybeeF4FR lite / CrazybeeF4FS lite V4.2`

* Please provide a link to the vendor specifications.  
https://www.happymodel.cn/index.php/2022/02/14/mobula6-v2-0-upgraded-flight-controller-crazybeef4fr-lite-crazybeef4fs-lite-v3-1-version-update-log/

**CLI dump or Unified Target**
* Please provide a CLI `dump` of **DEFAULTS** or link to unified target.

Unified target  
* https://github.com/betaflight/unified-targets/blob/master/configs/default/HAMO-CRAZYBEEF4FS.config
* https://github.com/betaflight/unified-targets/blob/master/configs/default/HAMO-CRAZYBEEF4FR.config

nerdCopter, 2024-04-08:
do you need Analog or MSP-Display-port?
[EmuFlight_0.4.2_CRAZYBEEF4FR_testing_builds.zip](https://github.com/emuflight/EmuFlight/files/14906115/EmuFlight_0.4.2_CRAZYBEEF4FR_testing_builds.zip)
  The hex with "HD" in the name requires unreleased Configurator: https://github.com/emuflight/dev-master/releases/tag/20240217.190950-cfg
  The "Analog" can you the released 0.4.1 version.
  
Seems to have no Black-box flash. is this correct?

please test/feedback any/all features applicable:
  - [Yes/No/Untested] hex Flash & subsequent USB connect
  - [Yes/No/Untested] Gyro 
  - [Yes/No/Untested] Accelerometer
  - [Yes/No/Untested] Is the Serial Count correct? (Ports tab UARTs)
  - [Yes/No/Untested] Motors
  - [Yes/No/Untested] Onboard ADC
  - [Yes/No/Untested] OSD (analog)
  - [Yes/No/Untested] Smart Audio controls
  - [Yes/No/Untested] Beeper/Alarm switch
  - [Yes/No/Untested] Barometer
  - [Yes/No/Untested] Blackbox  flash-chip/sdcard
  - [Yes/No/Untested] LED Strip
  - [Yes/No/Untested] GPS
  - [Yes/No/Untested] BlueTooth (if applicable)

nerdCopter, 2024-04-08:
deleting that build. it did not include FRsky drivers. will try again.

nerdCopter, 2024-04-08:
[EmuFlight_CRAZYBEEF4__BMI270_ICM42688P_Analog_needs_testing.zip](https://github.com/emuflight/EmuFlight/files/14906834/EmuFlight_CRAZYBEEF4__BMI270_ICM42688P_Analog_needs_testing.zip)

Please test and reply @alber70g . Thank you.

closed 2024-04-09 by nerdCopter