emuflight/EmuFlight #1029: HGLRC SPECTER F405V2 Target Needed

enterobotica, 2024-04-17:
**Vendor/Model**
HGLRC Specter F405 V2

* Please provide the target name.
HGLRCF405V2

* Please provide vendor and model of the FC.
HGLRC SPECTER F460 STACK

* Please provide a link to the vendor specifications.
https://www.hglrc.com/products/hglrc-specter-f460-stack-for-fpv-racing-drone-freestyle

**CLI dump or Unified Target**
https://github.com/betaflight/unified-targets/blob/master/configs/default/HGLR-HGLRCF405V2.config

nerdCopter, 2024-04-17:
Analog or HD?
Only the BMP270 version of the Baro will work (The DPS310 will not as we do not have working drivers). Do you you need magnetometer?

enterobotica, 2024-04-17:
I'm using ANALOG VTX (XF5804 pro 5.8 GHz)
BARO: BMP280
I have a magnetometer but it is not a priority [LIS2MDL](https://robu.in/product/smartelex-lis2mdl-triple-axis-magnetometer/?gad_source=1&gclid=CjwKCAjw5v2wBhBrEiwAXDDoJeadvV0-HbGnKr_JOi83X3rkENRu8VKbqEKi2pFXuPGkaMB3gxf7NhoCzDAQAvD_BwE) 
GPS: [HGLRC M100 MINI](https://www.hglrc.com/products/hglrc-m100_mini-gps)

Thanks for your response, I appreciate your effort :)

nerdCopter, 2024-04-17:
please test [EmuFlight_0.4.2_HGLRCF405V2_Build_dbcc5821a.hex.zip](https://github.com/emuflight/EmuFlight/files/15015387/EmuFlight_0.4.2_HGLRCF405V2_Build_dbcc5821a.hex.zip)

  - [Yes/No/Untested] hex Flash & subsequent USB connect
  - [Yes/No/Untested] Gyro 
  - [Yes/No/Untested] Accelerometer
  - [Yes/No/Untested] Is the Serial Count correct? (Ports tab UARTs)
  - [Yes/No/Untested] Motors
  - [Yes/No/Untested] Onboard ADC
  - [Yes/No/Untested] OSD (analog)
  - [Yes/No/Untested] Smart Audio or Tramp controls
  - [Yes/No/Untested] Beeper/Alarm switch
  - [Yes/No/Untested] Barometer
  - [Yes/No/Untested] Blackbox  flash-chip/sdcard
  - [Yes/No/Untested] LED Strip
  - [Yes/No/Untested] GPS
  - [Yes/No/Untested] BlueTooth (if applicable)

nerdCopter, 2024-04-18:
let me know results so that i can merge/release this target.

enterobotica, 2024-04-18:
I installed the hex file today, I got all motors spinning but still figuring out how to change motor order in CLI. My board is upside down as it is a 3.5 inch cinewhoop. Here is what I could see after my install:

[**Yes**/No/Untested] hex Flash & subsequent USB connect
[**Yes**/No/Untested] Gyro
[**Yes**/No/Untested] Accelerometer
[**Yes**/No/Untested] Is the Serial Count correct? (Ports tab UARTs)
[**Yes**/No/Untested] Motors
[Yes/No/**Untested**] Onboard ADC
[Yes/No/**Untested**] OSD (analog)
[Yes/No/**Untested**] Smart Audio or Tramp controls
[Yes/No/**Untested**] Beeper/Alarm switch
[Yes/No/**Untested**] Barometer
[Yes/No/**Untested**] Blackbox flash-chip/sdcard
[**Yes**/No/Untested] LED Strip
[Yes/No/**Untested**] GPS
[Yes/No/**Untested**] BlueTooth (if applicable)

I'll keep you posted with further updates.

nerdCopter, 2024-04-18:
https://www.youtube.com/watch?v=4G0SZ4jUVw0
https://www.youtube.com/watch?v=RMzeTf5mBeE

nerdCopter, 2024-04-18:
actually, might just need Gyro/Accel FLIP. i would try such before motor resources.
example, i dont know which one
[image]

enterobotica, 2024-04-18:
I wasn't able to get the right motor mapping with sensor alignment, went ahead with remapping in CLI and it worked great. Had my first flight, filtering needs to be done better now, do we have any resources to get it done right?

Also,
How to know if Barometer is working?
How to check onboard ADC?

[EMUF_cli_20240418_231122.txt](https://github.com/emuflight/EmuFlight/files/15028729/EMUF_cli_20240418_231122.txt)
[Yes] hex Flash & subsequent USB connect
[Yes] Gyro
[Yes] Accelerometer
[Yes] Is the Serial Count correct? (Ports tab UARTs)
[Yes] Motors
[Untested] Onboard ADC
[Yes] OSD (analog)
[Yes] Smart Audio or Tramp controls
[Yes] Beeper/Alarm switch
[Untested] Barometer
[Yes] Blackbox flash-chip/sdcard
[Yes] LED Strip
[Yes] GPS
[Untested] BlueTooth (if applicable)

nerdCopter, 2024-04-18:
This is my tuning over-view: https://hackmd.io/@nerdCopter/HkZjwid9q , maybe a little outdated. i've been setting my filters, then setting 1 single dynamic-filter notch count with a fairly narrow Q on gyro and a littler wider on D and pretty much good to go in almost all cases.

If you getting Baro data, good enough.
[image]

If you getting voltage data, good enough. LiPo plugged should be near actual. may need tuning. 
[image]

nerdCopter, 2024-04-18:
Join Emuflight discord for tuning help https://discord.gg/BWqgBg3 

i have a 5" SourceOne V5 w/ icm42688p that flies near perfect with this:
[image]
[image]
[image]

enterobotica, 2024-04-18:
[Yes] Barometer
[Yes] Onboard ADC

Thanks for your prompt support nerdcopter, for all the resources, I'll follow these and let you know how it pans out.

Please let me know if I can still share updates or this thread needs to be closed.

nerdCopter, 2024-04-18:
i will merge the target and close this ticket, thank you kindly!

PR #1031 (2024-04-18): [target] add HGLRCF405V2
resolves #1029
commit 4b4704bc (2024-04-18): [target] add HGLRCF405V2 (#1031)

resolves #1029
commit 921c5c41 (2024-04-19): [target] add HGLRCF405V2 (#1031)

resolves #1029
closed 2024-04-18 by nerdCopter