emuflight/EmuFlight #1066: JHEMCU F745 AIO target with MSP functionality.

DutchFPV, 2024-05-23:
**Vendor/Model**
* Please provide the target name. e.g. `JHEF745`
* Please provide vendor and model of the FC. -JHEMCU F745 AIO- GHF745AIO
* Please provide a link to the vendor specifications.
https://www.jhemcu.com/e_productshow/?66-GHF745AIO-40A-66.html

**CLI dump or Unified Target**
* Please provide a CLI `dump` of **DEFAULTS** or link to unified target.
https://github.com/betaflight/unified-targets/blob/master/configs/default/JHEF-JHEF745.config

Hoping to get this with MSP functionality for use with HDZero.  Not sure if that has all been merged yet or not.

DutchFPV, 2024-05-23:
> here you go, i had to remove motor 7 and 8 for it to compile.
> EmuFlight_0.4.3_JHEF745_Build_94e9f5a2a.hex.zip
>
> please test and feedback. i'm okay with merging without motor 7 and 8 because they are no associated timers defined for them in BF.
>
> you have to use latest configurator here: https://github.com/emuflight/dev-master/releases/tag/20240501.162228-cfg

Thank you!  Its for a friend but I'll def confirm working and forward any feedback he has.

Derek C. Donohue ***@***.***

DutchFPV, 2024-05-26:
Got the following feedback from my friend after testing:
-Vtx control not working (vtx always switches to R1, not controllable through elrs vtxadmin) 
-Sometimes a motor won't spin when armed normally, and when trying to spin individual motors (either through emu config or using turtle mode switch) a motor often won't spin, then that esc reboots.

Let me know if you need any further info.  Thanks!

DutchFPV, 2024-08-28:
Ok reviving this after some further testing. We've established that the BF 4.4 target works with defaults applied.  Upon further review it looks like the defaults file I gave you changed since BF 4.2.  Here is a CLI dump from tested-working BF 4.4 target with custom defaults applied.  Can we give this a try?  Let me know if you need any more info
[btfl_cli_jhef745_440.txt.pdf](https://github.com/user-attachments/files/16788140/btfl_cli_jhef745_440.txt.pdf)
 
(BTW, we changed the file extension to .pdf to trick Facebook.  But this will open normally as a text file.)

nerdCopter, 2024-08-28:
there is a v1 and a v2, do you know which id proper for the hardware?
`JHEF745`   `JHEF745V2`

nerdCopter, 2024-08-28:
here is `JHEF745V2` minus motor 7 and 8 which did not have timers defined.
[EmuFlight_0.4.3_JHEF745V2_Build_0f9c0f77a.hex.zip](https://github.com/user-attachments/files/16788743/EmuFlight_0.4.3_JHEF745V2_Build_0f9c0f77a.hex.zip)

DutchFPV, 2024-08-28:
The BF target he's using is for V1.

nerdCopter, 2024-08-29:
i reviewed your .txt side-by-side, i see no differences to what is in master https://github.com/emuflight/EmuFlight/tree/master/src/main/target/JHEF745 :man_shrugging:

DutchFPV, 2024-08-29:
So you're saying what's in the BF dump matches the Emu master?  That's what I was trying to confirm.  Emu gave odd motor behavior and desyncs but the drone flew fine on BF.  Guess it could have been a flash error or something else.  Will try again with the master from the configurator and confirm.  Thanks.

nerdCopter, 2024-08-29:
i suppose you could get `timers` and `dma` from emu and we can re-compare.  otherwise the all the pins are identical.

nerdCopter, 2024-08-29:
for what it's worth BF devs and community have noted JHEMCU as typically low quality.

DutchFPV, 2024-08-29:
We'll see what happens with another test.  If it still has issues I'll dump timers and dma.  Thanks!

DutchFPV, 2024-08-29:
So same thing.  In motors tab all motors spin together, but motors 3 and 4 desync if you try to spin them individually.  These same motors are giving erratic behavior in flight.  I think I found dma, but how can I query timers?  I'll upload both for you.

nerdCopter, 2024-08-30:
dang, looks like Emu does not have the timers command :( :face_with_head_bandage:

nerdCopter, 2024-08-30:
can try `set dshot_burst = ` `ON`|`OFF` , but probably no help.  not sure where to go from here.

DutchFPV, 2024-09-26:
Just a final update that this target is fully operational.  Dshot burst was off by default.  Turned it on and everything works great. HDzero support seems pretty solid too.

nerdCopter, 2024-09-26:
thanks, i merged [the fix](https://github.com/emuflight/EmuFlight/pull/1087) and updated the .hex in releases.

closed 2026-05-02 by nerdCopter