emuflight/EmuFlight #1070: foxeer f722 v4 some bugs

superLi88, 2024-06-22:
Guys, why do I use foxeer f722 v4 to write emuflight, connect to the ground station, and then swing my plane, the direction of the gyroscope above the ground station is reversed, I have set from cw0 '-270' in the setting again, but it is still not working, what is the reason, a single flight control bug?

I don't have that problem with bf

nerdCopter, 2024-06-24:
> 
[image]

did you solve? i dont know about ground-station.  emulfight is a butterflight fork, which is essentially a betaflight 3.5.7 fork.

nerdCopter, 2024-06-24:
okay i see #1071 , closing #1070.

closed 2024-06-24 by nerdCopter