emuflight/EmuFlight #1071: foxeer f722 v4 gyro bug

superLi88, 2024-06-23:
**Describe the bug**
A clear and concise description of what the bug is.
The gyroscope attitude is reversed
**To Reproduce**
Steps to reproduce the behavior:
I used the firmware version 0.4.3, after connecting and entering the ground station, calibrating the accelerometer and resetting the Z-axis, moving the fc the position I moved in reality is completely opposite to the position I moved in the computer, and I adjusted the direction of the gyroscope from cw0 '-270' in the setting, is this a single bug corresponding to a single flight control?
**Expected behavior**
A clear and concise description of what you expected to happen.
Corrected gyroscope attitude
**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)
# DUMP

# version
# Betaflight / STM32F7X2 (S7X2) 4.3.2 Nov 28 2022 / 07:30:19 (60c9521) MSP API: 1.44

# config: YES

# start the command batch
batch start

board_name FOXEERF722V4
manufacturer_id FOXE

# name: -

# resources
resource BEEPER 1 A04
resource MOTOR 1 A09
resource MOTOR 2 A08
resource MOTOR 3 C09
resource MOTOR 4 C08
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B07
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A15
resource SERIAL_TX 1 B06
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 B07
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
[... 1073 lines cut ...]
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 67
set tpa_rate = 65
set tpa_breakpoint = 1350
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998
set roll_level_expo = 0
set pitch_level_expo = 0

# end the command batch
batch end

# 
**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];  FOXEER F722 V4   with foxeer.com
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.
My receiver is connected to r1t1,and My image transmission is connected to y5 
**Additional context**
Add any other context about the problem here.

nerdCopter, 2024-06-24:
does it work on betaflight or inav correctly?
the default for the FC is CW270.  if you orient the FC differently, you must add it to the CW270.
i also don't see `align_acc = ` in your dump.

superLi88, 2024-06-25:
> does it work on betaflight or inav correctly?
> the default for the FC is CW270. if you orient the FC differently, you must add it to the CW270.
> i also don't see `align_acc = ` in your dump.

Yes, my power cord is front-facing, so I set cw 90 on betaflight and it works normally. I don't know why it doesn't work normally in emu. Do you have a convenient way to contact me? I want to solve this problem quickly

nerdCopter, 2024-06-25:
Discord https://discord.gg/BWqgBg3
is this what you set?
[image]

nerdCopter, 2026-05-02:
@coderabbitAI, so the solution is to set BOTH as per screenshot? If only gyro set, then breakage?

[image]