emuflight/EmuFlight #1079: Target Request: NBD_INFINITYAIO255

BeauBrewski, 2024-08-17:
**Vendor/Model**

`NBD_INFINITYAIO255`

NewBeeDrone Infinity AIO 255 Flight Controller F7

https://newbeedrone.com/products/infinity-aio-255

**CLI dump or Unified Target**

https://github.com/betaflight/unified-targets/blob/master/configs/default/NEBD-NBD_INFINITYAIO255.config

BeauBrewski, 2024-08-17:
***Note: Target is the single gyro, MPU6000 version of NBD_INFINITYAIOV2PRO**

nerdCopter, 2024-08-17:
Will do. Officially not merged on BF, but i think still do-able.

nerdCopter, 2024-08-19:
needs full testing: 
[EmuFlight_0.4.3_NBD_INFINITYAIO255_Build_08ea01bc8.hex.zip](https://github.com/user-attachments/files/16662533/EmuFlight_0.4.3_NBD_INFINITYAIO255_Build_08ea01bc8.hex.zip)

nerdCopter, 2024-08-19:
weird layout, looks like traditional 30x30 AIO, but it's 25.5. so is that for diagonal mounting? then motor wires askew?

nerdCopter, 2024-08-19:
proprietary plug? https://cdn.shopify.com/s/files/1/1126/9610/files/Infinity_AIO_255-_-v1.1-20231113__1_4.jpg
hd plug i presume: https://newbeedrone.com/products/infinity-aio-255

BeauBrewski, 2024-08-19:
Recommended for 3"-4" ultralight? Looks like it'd be a clean build, but yeah, either use adaptor plate, a frame with square 25.5x25.5, or have skewed wires. I didn't even pay that close of attention to the FC layout, was more annoyed that NBD didn't have target listed on their own page.

closed 2024-11-12 by nerdCopter